gyroflow/gyroflow #411: Not work in Ubuntu 22.04

ser812, 2022-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.1.0

### What operating system are you using?

Ubuntu 22.04

### What GPU are you using?

Intel(R) Core(TM) i5-3337U CPU @ 1.80GHz

### What happened?

Crash

### Relevant log output

```shell
sergey@sergey-700T1C:~/Загрузки/Gyroflow$ ./gyroflow 
MDK 0.14.2 (git e2a85c2) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.5 GCC4.2.1; libc++14000; 00:00:12 Jun 13 2022
07:22:57 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
07:22:57 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
07:22:57 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
07:22:57 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
07:22:57 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
07:22:57 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
07:22:59 [INFO] Lens profiles directory: "/home/sergey/Загрузки/Gyroflow/camera_presets"
07:22:59 [INFO] Loaded 3333 lens profiles in 64.601ms
07:22:59 [INFO] Latest version: v1.1.0, current version: 1.1.0 (gh570)
07:23:00 [INFO] Downloading lens profile "Runcam_ThumbPro_1080p_4by3_1920x1440-60.00fps.json"
07:23:00 [INFO] Downloading lens profile "Runcam_ThumbPro___2.7k_16by9_2704x1520-59.94fps.json"
thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: Unable to get platform id list after 10 seconds of waiting.', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:50:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Rayon: detected unexpected panic; aborting
Crashdump written to /home/sergey/Загрузки/Gyroflow/44fa243f-c087-47fa-45ec9e9a-33e25b05.dmp
Аварийный останов (стек памяти сброшен на диск)
```

ser812, 2022-07-25:
[dump.zip](https://github.com/gyroflow/gyroflow/files/9179060/dump.zip)

AdrianEddy, 2022-07-25:
Did you install all recommended ackages mentioned in the README? Especially OpenCL.

AdrianEddy, 2022-07-25:
The error comes from OpenCL, which is GPU dependent. Which GPU do you have?
Did you try to install `pocl-opencl-icd`?

ser812, 2022-07-25:
integrated GPU

after install pocl-opencl-icd it work

closed 2022-07-25 by ser812