gyroflow/gyroflow #412: Interface is not loading on Windows 11

gurfing, 2022-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.1

### What operating system are you using?

Windows 11 Pro 21H2

### What GPU are you using?

NVIDIA RTX 3050 Ti, Intel UHD

### What happened?

Program goes to background after run without any interface.

### Relevant log output

```shell
20:25:51 [WARN] QQmlApplicationEngine failed to load component
20:25:51 [WARN] qrc:/src/ui/main_window.qml: No such file or directory
20:25:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["NVIDIA GeForce RTX 3050 Ti Laptop GPU"]
20:25:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) UHD Graphics"]
20:25:52 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce RTX 3050 Ti Laptop GPU
20:25:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
20:25:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:51] gpu_type = NVIDIA
```

AdrianEddy, 2022-07-31:
Update your GPU drivers

gurfing, 2022-08-01:
All drivers are the latest + all windows updates intalled.

AdrianEddy, 2022-08-01:
Ok, did you try latest dev build? https://gyroflow.xyz/devbuild/?autodownload

gurfing, 2022-08-01:
Yes. I think the main problem is in:
```
20:25:51 [WARN] QQmlApplicationEngine failed to load component
20:25:51 [WARN] qrc:/src/ui/main_window.qml: No such file or directory
```
I looked through logs from other issues in this gitlab repo and never saw this.

AdrianEddy, 2022-08-01:
that's weird, I don't see a way for the embedded files to be affected by anything external like your system and it works for everyone else, so I have no idea why it would happen for you. As a workaround, you can try to download the code from the repository (green button "Code" -> Download ZIP), and copy the `ui` folder from `src/` from the repo, to the same folder as your downloaded Gyroflow.exe

gurfing, 2022-08-02:
That helped, thanks!

closed 2022-08-02 by gurfing