gyroflow/gyroflow #414: AutoSync fails: Optical flow only returns Z axis, X and Y are either 0 or 1

ScissorHill, 2022-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.1.0 (gh570)

### What operating system are you using?

Ubuntu 22.04

### What GPU are you using?

Nvidia GeForce GTX 1060

### What happened?

[image]

I''m trying to stabilize video shot with Nikon Coolpix P950, using a Runcam Thumb as a gyro logger.

Since the camera profile doesn't exist, I've run the calibration myself (video and JSON file [here](https://drive.google.com/drive/folders/1cVo7WMnNrDsbKAgE00Mv30QHjsO7DX1L?usp=sharing)).

However, when I hit Auto Sync, Optical Flow graph looks weird: X and Y are not continuous but discrete:

[image]

Obviously, the stabilization fails.

The test video, Runcam Thumb video and .gcsv files are [here](https://drive.google.com/drive/folders/1hm_usQS-OF18c-Ho4q3RzEggN5b7P2WP?usp=sharing).

As a sanity check, the Runcam Thumb video stabilization works fairly well.

Any help in getting the stabilization to work would be appreciated.

### Relevant log output

```shell
MDK 0.14.2 (git e2a85c2) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.5 GCC4.2.1; libc++14000; 00:00:12 Jun 13 2022
04:38:52 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
04:38:52 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
04:38:52 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
04:38:52 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
04:38:52 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
04:38:52 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

04:38:53 [INFO] Lens profiles directory: "/home/guru/src2/Gyroflow-linux64/Gyroflow/camera_presets"
04:38:53 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["NVIDIA GeForce GTX 1060 with Max-Q Design"]
04:38:53 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1060 with Max-Q Design
04:38:53 [INFO] Loaded 3335 lens profiles in 46.333ms
04:38:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
04:38:54 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src/rendering/mod.rs:51] gpu_type = NVIDIA
04:38:54 [INFO] Latest version: v1.1.0, current version: 1.1.0 (gh570)
04:38:55 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
04:39:28 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
04:39:28 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "thread.video" ""
04:39:28 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
04:39:28 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
04:39:28 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
04:39:28 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.MMAL.mmal"
04:39:28 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
04:39:28 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fde2cdb2440) QSize(1146, 645)
04:39:28 [DEBUG] (17) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
04:39:28 [INFO] surface size: 1280x720
04:39:28 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
04:39:28 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fde2c227a40) QSize(1280, 720)
[/root/.cargo/git/checkouts/telemetry-parser-19821e7ac6d09f51/74ed178/src/util.rs:348] reading_duration = (
    None,
    None,
    None,
)
04:40:07 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_CUDA 0
04:40:07 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_CUDA (cuda) with format Some(AV_PIX_FMT_CUDA)
04:40:28 [INFO] Setting offset at 4754.7500: 2554.7459 (cost 2.0148)
04:40:28 [INFO] Setting offset at 14264.2500: 4669.3298 (cost 2.1595)
```

AdrianEddy, 2022-08-03:
This is likely because it's an extreme focal length, which current  lens model doesn't support well. It will get better once I implement #150

ScissorHill, 2022-08-04:
Thanks. I expect this to be very useful for bird/wildlife videographers, so eagerly looking forward to the implementation. 

Question 1: Not sure I understand much of the discussion in [#150](https://github.com/gyroflow/gyroflow/issues/150);  Would the calibration work with the legacy Python code?

Question 2: While running the calibration, the colored lines  seemed to mostly be missing the checkerboard most of the time:

AdrianEddy, 2022-08-04:
1. No, legacy python code uses the same model
2. Is this linux? it's possible is a display bug on linux, but it shouldn't affect the calibration. I tested the calibration with your file and the lines were fine on Windows, but due to extreme focal length it was still unusable

ScissorHill, 2022-08-04:
2. Yes indeed, Ubuntu 22.04.

AdrianEddy, 2022-12-01:
I spent a lot of time researching and trying to understand this scenario and it's a difficult problem. 

First of all, stabilization of large focal lengths works fine as long as you have solid gyro data and accurate sync.

The problem is in the optical flow and determining the camera rotation and translation at these large focal lengths. With a telephoto lens (eg. over 500mm) it's nearly impossible to distinguish between camera rotation and translation, and most algorithms fail to do that.

There are some algorithms that assume 0 translation and just focus on rotation, which is never true for handheld footage, but in this case it's more suitable than the generic algorithm that gets confused and returns completely invalid result.

Latest dev build of Gyroflow adds a way to select `Pose method`, and the rotation-only is `Almeida` which you can select and use with optical flow. This one produces *some* results that can be usable.

However, I was unable to stabilize your video:
1. Runcam Thumb is not a great quality device
2. How was the Thumb mounted? Was it absolutely rigidly connected to the camera? Was it absolutely perfectly aligned with the camera lens? At these focal lengths these little details matter a lot
3. How long was the delay between starting the Thumb and Nikon recordings? You need to help the sync algorithm by providing this `Rough gyro offset`
4. Possibly take extra time to record better calibration footage, try more different angles (you'll need a lot of physical space for that, but it will help the end result)

You can stabilize such footage, but you need to take extra care on the hardware side before attempting sync in Gyroflow.

Since the new algorithms were added to Gyroflow, I'm closing this issue as there's not much more that can be done on the software side.

Latest dev build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-12-01 by AdrianEddy