gyroflow/gyroflow #415: Feature request: Stitch separated gopro footages

temp-64GTX, 2022-08-02:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

It's not a secret that gopro have a limitation from 80-s, so that maximum footage size is limited to 3.7 Gb (about 4-5 minutes). So if you want to shoot a continuous 15-minutes run - gopro will break it to 4 picies.
And this is causing obvious stabilisation problems. Files are separated, gyro-data are separated. If we will stabilizing them one by one - then we will have a nasty step in final video, when one footage ends, and second is starts. 
I hope that i explained the point.
So it will be a cool option - to stitch footages together, so that gyroflow will interprete all pieces of video and gyro as one continious footage.

[image]

AdrianEddy, 2022-08-02:
There's already https://github.com/rubegartor/ReelSteady-Joiner for that

AdrianEddy, 2022-08-24:
Done in 06d723d44df9edd18d01d69ce4864ba82ac8410c and https://github.com/gyroflow/mp4-merge/
Please test in latest dev build and let me know if it worked as expected in Gyroflow:  https://gyroflow.xyz/devbuild/?autodownload
When loading first file from the sequence, Gyroflow should detect and ask to merge.

closed 2022-08-24 by AdrianEddy

estebanavv, 2022-09-22:
This works great for video, is there a way to stitch .gcsv files?