gyroflow/gyroflow #426: Allow reading of DJI FPV Drone Gyro Data

CarbonFivrr, 2022-08-12:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi, I was finally able to pull the gyro data from the DJI FPV Drone but when I tried to upload it into GyroFlow it was unable to read the format. I have attached the flight log here. Hopefully this helps and I would love to see this feature added. I was so excited when I found out about this program!
[DJI FPV Drone Flight Log.csv](https://github.com/gyroflow/gyroflow/files/9313105/DJI.FPV.Drone.Flight.Log.csv)

AdrianEddy, 2022-08-12:
Do you have a corresponding video file? Where is this log from?

CarbonFivrr, 2022-08-12:
It is from the native gyro data from the DJI FPV drone. And yes, I have the corresponding video but it won't let me attach it here because its too big

AdrianEddy, 2022-08-12:
Use WeTransfer or Google drive. But how did you export that data? Is it a flight log from the app?

CarbonFivrr, 2022-08-12:
Yep, its a flight log from the app and then I used a program called CSVview to convert the .DAT file into a CSV.  

Here is the link to the video: https://we.tl/t-Gg4LMPrtTd

azuretimm, 2022-08-27:
> Yep, its a flight log from the app and then I used a program called CSVview to convert the .DAT file into a CSV.
> 
> Here is the link to the video: https://we.tl/t-Gg4LMPrtTd

I assume it is this program? https://datfile.net/CsvView/downloads.html
How did you extract the .DAT files? Do you have to connect the phone to the goggles while flying and have the DJI Fly app running to log the data? Is there a way to pull the .DAT files from the drone directly without connecting the phone? Thank you.

CarbonFivrr, 2022-08-27:
Yea so you just have to connect the phone to the goggles and go into the DJI Fly app click on Profile and then the more button that’s under your account username. From there it takes you to the “Flight Data Center” and at the top right corner, there’s a button that looks like a clock. You click that and download the flight data from the corresponding flights. From there it should download them to your phone. For IOS it should make a folder in the “Files” app called “DJI Fly”. Inside that folder find the folder that says “Flight Records” and all the stuff is in there. 

You can pull the logs straight from the goggles without having the drone connected but you do have to connect the phone to the goggles. From what I’ve read online, you could connect the drone to a computer and use the DJI app for the computer but then the files come out encrypted. For some reason if you do it this way, it doesn’t encrypt them.

mattbratt, 2022-09-08:
Wow, this is great news to know that the data is there. Just need to find a way to ingest it into gyroflow. Thanks! I will look forward to any progress on this, as so many have the DJI FPV drone that would love to use gyroflow to stabilize the footage.

AdrianEddy, 2022-09-08:
> Wow, this is great news to know that the data is there. Just need to find a way to ingest it into gyroflow. Thanks! I will look forward to any progress on this, as so many have the DJI FPV drone that would love to use gyroflow to stabilize the footage.

don't get too excited though, that data may be unusable. I didn't check it yet, but if the gyro is not on the gimbal directly behind the camera (and it isn't), then it may not be possible to get good results with that data

BudWalkerJava, 2022-09-12:
> > Wow, this is great news to know that the data is there. Just need to find a way to ingest it into gyroflow. Thanks! I will look forward to any progress on this, as so many have the DJI FPV drone that would love to use gyroflow to stabilize the footage.
> 
> don't get too excited though, that data may be unusable. I didn't check it yet, but if the gyro is not on the gimbal directly behind the camera (and it isn't), then it may not be possible to get good results with that data. Also sampling frequency may be too low.

Is there some documentation that specifies the format of the .csv containing the gyro data?

AdrianEddy, 2022-09-12:
> Is there some documentation that specifies the format of the .csv containing the gyro data?

Sure, https://docs.gyroflow.xyz/logging/gcsv/

wzqvip, 2022-10-05:
I wrote a simple python file to convert the data format. But I don't know much about the DJI's data format, so I left some parameters in the program. Maybe we can fix it out.

(.py is not supported by github.)
[transfer.py.txt](https://github.com/gyroflow/gyroflow/files/9713360/transfer.py.txt)

unlimitedit, 2022-10-21:
To try to help progress work on Gyroflow for the DJI FPV Drone, I have shared on [OneDrive](https://1drv.ms/u/s!Ak61-4OMONxunPwEgi7_rCgX9F0Kfw?e=N4zpKC) the data captured from 5 flights along with the drone and goggle video. No internal stabilisation or distortion correction was used for the flights.

Flight 1    N Mode    4:3
Flight 2    N Mode    16:9
Flight 3    S Mode    16:9
Flight 4    S Mode     4:3
Flight 5    M Mode    4:3

The drone is always in 16:9 4K60 whilst the goggles change to 4:3 as stated.

The CSV folder contains CSV files derived from the unencrypted Goggles V2 DAT files (in the DAT folder) processed through [csvView](https://datfile.net/CsvView/intro.html). It might be possible to match the gyro, accel and mag data in the CSV files to the fields described in the [Gyroflow .gcsv reference](https://docs.gyroflow.xyz/logging/gcsv/).

Perhaps...

IMU_ATTI(0):gyroX | IMU_ATTI(0):gyroY | IMU_ATTI(0):gyroZ | IMU_ATTI(0):accelX | IMU_ATTI(0):accelY | IMU_ATTI(0):accelZ | IMU_ATTI(0):magX | IMU_ATTI(0):magY | IMU_ATTI(0):magZ

...to match

gx,gy,gz,ax,ay,az,mx,my,mz

I'm unsure as to what data should populate _t, tscale, gscale, ascale and mscale_ though.

I recognise that, as @AdrianEddy has suggested, the data may be unusable as the gyro is not on the gimbal directly behind the camera and the sampling frequency may be too low. However, there may be camera tilt data contained within the CSV that can be used to calculate the offset in N and S modes. For the last flight I used a camera tilt of 29 degrees.

BudWalkerJava, 2022-10-24:
t (time) would be Clock:offsetTime. tscale is seconds.

gscale is degs/sec

ascale is "g" ; E.g., IMU(0):accelZ is 1.0 when the drone is sitting on  level ground. IMU_ATTI(0):accelX and IMU_ATTI(0):accelY are  both 0.0

mscale doesn't really have a unique value. That's because the raw sensor data is scaled and biased depending on the latest compass calibration. But, IMU_ATTI(0):magX | IMU_ATTI(0):magY | IMU_ATTI(0):magZ are consisten with one another. If the drone is pointed into the geomagnetic field then IMU_ATTI(0):magY |and IMU_ATTI(0):magZ will be 0.0. Haven't looked at the gyroflow code but I think the magnetometer data are used to compute a direction vector in the geomagnetic frame of reference? And it doesn't matter what the mscale is?

AdrianEddy, 2022-10-24:
mscale doesn't matter and magnetometer is optional. Currently no camera has magnetometer data that I know of. It is used for IMU fusion if present, but it works fine without it as well

BudWalkerJava, 2022-10-24:
Is there an option to use IMU(0):roll, IMU(0):pitch and IMU(0):yaw that is the result of the result of the flight controller's fusion algorithm?

AdrianEddy, 2022-10-24:
Yes, quats can be used directly too, but not from .gcsv file. Custom parser needs to be implemented in telemetry -parser https://github.com/AdrianEddy/telemetry-parser

AdrianEddy, 2022-11-09:
Like I suspected, the sampling frequency is too low, just 20Hz, which is unusable for any stabilization. On top of that, I couldn't find any information about camera angle in the data, so it would only work if you fly in manual and have the camera tilt set to a fixed angle.

I added basic reading of these files here https://github.com/AdrianEddy/telemetry-parser/blob/master/src/dji/csv.rs it has correct timestamps and IMU orientation, but it's unusable to stabilize the video

closed 2022-11-09 by AdrianEddy

unlimitedit, 2022-11-09:
Thanks for your analysis @AdrianEddy . At least we know it's a dead end for sure now.