gyroflow/gyroflow #430: Quitting/crashing after clicking 'export'

Iroquoi5pliskin, 2022-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.1.0

### What operating system are you using?

macOS Catalina 10.15.7

### What happened?

Gyroflow's worked fine for me in the earlier version, but just updated to 1.1.0 & now after I select 'export' after importing video, setting lens, importing motion data, auto-syncing. All other settings left standard.

As soon as I press export, it looks like it's going to start rendering - but freezes & closes the application instead.

AdrianEddy, 2022-08-12:
Please include gyroflow.log (Open package contents->Contents->MacOS) and try the latest dev build https://gyroflow.xyz/devbuild/?autodownload

Iroquoi5pliskin, 2022-08-13:
Thanks.

I've downloaded that version. Still getting the same issue. Log attached.

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9331136/gyroflow.log)

AdrianEddy, 2022-08-17:
Can you please test if this also happens when you uncheck `Advanced` -> `Use GPU decoding`
Please also try to choose different devices in `Advanced` -> `Device for video processing` and `Export` -> `Device for rendering` and let me know

Iroquoi5pliskin, 2022-08-19:
Absolutely fantastic.

The GPU decoding setting still crashed the program, but changing the device for rendering setting sorted the crashing issue.

All working fine now. Thanks so much for your help.

closed 2022-08-19 by AdrianEddy