gyroflow/gyroflow #431: Gyroflow crash when export video

Pernetta, 2022-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

 v1.1.0

### What operating system are you using?

Windows 10

### What GPU are you using?

AMD Fx 6200 , AMD R7 370 video card

### What happened?

Hi all

When i try to export the video, the program crash to window, how fix it?

Thanks

AdrianEddy, 2022-08-15:
Run Gyroflow_with_console.bat and show me the console output after it crashes

Pernetta, 2022-08-15:
How run console bat?

Enviado do meu Galaxy

-------- Mensagem original --------

Data: 15/08/2022 10:58 (GMT-03:00)
Para: gyroflow/gyroflow ***@***.***>
Cc: Pernetta ***@***.***>, Author ***@***.***>

AdrianEddy, 2022-08-15:
just double click on that file `Gyroflow_with_console.bat`

Pernetta, 2022-08-15:
Hi,

Now when i drop the video , the program crash!!

15:56:28 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:56:28 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:56:28 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:56:28 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:56:28 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
15:56:28 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
15:56:30 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\fbvel\\Downloads\\Gyroflow\\camera_presets"
15:56:30 [INFO] Latest version: v1.1.0, current version: 1.1.0
15:56:30 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("AMD Accelerated Parallel Processing") ["Bonaire"]
15:56:30 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. Bonaire
15:56:30 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_D3D11VA
15:56:30 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_D3D11VA
[src\rendering\mod.rs:51] gpu_type = AMD
15:56:30 [INFO] Loaded 3335 lens profiles in 298.827ms
15:56:39 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
15:57:25 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:57:25 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "thread.video" ""
15:57:25 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
15:57:25 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
15:57:25 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:57:25 [DEBUG] (11) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
15:57:25 [DEBUG] (11) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x264cf1ca820) QSize(1164, 655)
15:57:25 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"
15:57:25 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
15:57:25 [INFO] surface size: 1280x720
15:57:26 [ERROR] OpenCL error:

################################ OPENCL ERROR ###############################

Error executing function: clCreateCommandQueue

Status error code: CL_OUT_OF_HOST_MEMORY (-6)

Please visit the following url for more information:

https://www.khronos.org/registry/cl/sdk/1.2/docs/man/xhtml/clCreateCommandQueue.html#errors

#############################################################################

15:57:26 [DEBUG] (11) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "AMD Radeon (TM) R7 360 Series", vendor: 4098, device: 26207, device_type: DiscreteGpu, backend: Vulkan }
15:57:26 [DEBUG] (11) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
15:57:26 [DEBUG] (11) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x264e877c490) QSize(1280, 720)
Crashdump written to C:\Users\fbvel\Downloads\Gyroflow/f5d0a78b-1286-415c-8270-45d9c8f712d0.dmp
Pressione qualquer tecla para continuar. . .

AdrianEddy, 2022-08-15:
Before you load a video, try changing the "Device used for processing" in the Advanced tab

AdrianEddy, 2022-08-15:
and make sure you have latest GPU drivers installed https://drivers.amd.com/drivers/radeon-software-adrenalin-2020-22.6.1-win10-win11-64bit-legacyasics-june23-2022-legacy.exe

Your GPU is probably too old to support GPU processing so if you have latest drivers and none of the GPU options in "Advanced" tab work, then you have to use CPU-only processing and rendering, make sure to select "CPU only" and uncheck "Use GPU encoding"

closed 2022-08-15 by AdrianEddy

Pernetta, 2022-08-15:
Hi, i update my video card driver end now its working good!!
Thanks for help.

Enviado do meu Galaxy

-------- Mensagem original --------

Data: 15/08/2022 13:11 (GMT-03:00)