gyroflow/gyroflow #433: This application failed to start because no Qt platform plugin could be initialized

Magadeev, 2022-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.1.0

### What operating system are you using?

Windows 10 x64

### What GPU are you using?

Intel 4400

### What happened?

does not start
[image]

### Relevant log output

```shell
12:37:54 [INFO] Could not load the Qt platform plugin "windows" in "" even though it was found.
12:37:54 [ERROR] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: windows.

Для продолжения нажмите любую клавишу . . .
```

AdrianEddy, 2022-08-18:
Install https://aka.ms/vs/17/release/vc_redist.x64.exe and install all windows updates. Can you show me a screenshot of System info from control panel?

Magadeev, 2022-08-18:
vc_redist.x64.exe installed. The update cannot be installed because the corporate version of Windows
[image]

Magadeev, 2022-08-18:
1511 version =D

AdrianEddy, 2022-08-18:
Yeah so it's not supported, you have to upgrade the Windows version or use a different computer.

closed 2022-08-18 by AdrianEddy