gyroflow/gyroflow #438: Normalize speed

eortegaz, 2022-08-25:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Normalizing speed based on telemetry data. Automatic speed ramping to make speed constant!

AdrianEddy, 2022-08-25:
I believe this is the same as #142, it may be implemented in the future but it's not trivial and we need to do #31 first.
Please follow #142 for the progress

closed 2022-08-25 by AdrianEddy

eortegaz, 2022-11-03:
> I believe this is the same as #142, it may be implemented in the future but it's not trivial and we need to do #31 first. Please follow #142 for the progress

Thanks for your quick reply. Looks like #31 is now done. Speed normalization is still low-priority but I hope it's a feature we'll see in the future.