gyroflow/gyroflow #443: Blackmagic Footage direct rendering

mrnocreativity, 2022-08-27:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

First of all, thanks for the great work on the .braw/gyro support! It's incredible how fast the team has added support for this! Huge props to everyone involved.

From the documentation, I've read that it is currently not possible to render .braw videos from Gyroflow into stabilized videos directly. Only using the OpenFX workflow. I would really like it if it were possible to be able to render directly to ProRes footage. Is there any chance this will be worked on in the future? Or are there reasons why this will not/can not be supported in the future?

Thanks for your consideration and hard work!
[image]

AdrianEddy, 2022-08-28:
Ok so here's the situation:
BRAW playback was not something I could implement myself, but a lot of people asked about it. I did some research and it turned out it's possible to add it, but it would have to be implemented by the author of the playback library Gyroflow uses. This library is closed source, so I reached out to the author asking for that implementation and he said it's possible if I pay him for that work. I personally don't need .braw support but since a lot of people asked for it, I figured I can make it happen for them, if they raise the funds to pay for that developer's work.

I didn't want to make empty promises though, so I asked him to start working on it, we proved it worked, and I paid him for the initial implementation from my own money. Then I made a post on various groups on facebook ([1](https://www.facebook.com/groups/cinelifters/posts/909075150495706/), [2](https://www.facebook.com/groups/fpvtools/posts/1198088734371639/)), I shown a demo where this works, and I asked the community to fund this work if they have a need for it.

The cost of that developer work alone was $3000, not counting my part, and the raised amount after spreading the word across various groups and a few weeks was just under $1000.

That was the playback part, and that was working. In order to implement exporting to ProRes (or any exporting for that matter) I had to do a lot more work on my part and implement/rewrite a lot of code to make this happen. It's possible, but not something plug-and-play.
I initially planned to implement that, but once it turned out that there's not that much demand for it (a lot more people asked for it than actually chipped in), and I paid the rest from my own pocket, I decided it's not something I want to spend more weeks on implementing and I have better use of my time.

So for now, you can use Resolve with the plugin, and have all your braw control there. I decided it's better to spend more time improving the plugin than implementing the rendering in Gyroflow itself.

mrnocreativity, 2022-08-28:
Thanks for your elaborate explanation. I really appreciate that. 

I see where this decision is coming from. Too bad. I'd be willing to chip in for this functionality, obviously as it would benefit my commercial workflows too.

Thanks for your hard work and your support.