gyroflow/gyroflow #444: Gopro Hero 8 Linear footage starts with zoom in

B-nutze-RR, 2022-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.2 and up to gh711 tested but 1.1 is fine

### What operating system are you using?

win10

### What GPU are you using?

Nvidia

### What happened?

Hero a sample starts fully zoomed in and the quaternions graph shows a dip - in 1.1 the dip is not present
[image]
[image]
.

### Relevant log output

```shell
video:
https://disk.yandex.ru/i/-pfZh7_vVUnGsA
```

AdrianEddy, 2022-08-27:
@VladimirP1 I believe this is due to QuaternionConverter, can you take a look? If it's too hard to fix, we can disable quaternion converter when horizon lock is not being used

VladimirP1, 2022-08-28:
Should be fixed in #445

commit db7fe045 (2022-08-28): Merge pull request #445 from VladimirP1/master (Fixes #444)

apply full correction on first iteration in quaternionconvertor
closed 2022-08-28 by AdrianEddy