gyroflow/gyroflow #448: DJI AVATA Combining multiple files over 3.5GB

twtps0328, 2022-08-29:
Hi there,

If working with DJI AVATA split files , How can we Combining .mp4 or .gyroflow ?

AdrianEddy, 2022-08-29:
If your file is split at 6:00, Gyroflow should detect it automatically if you load the first file.
If it's not, you can merge the files manually using https://github.com/gyroflow/mp4-merge (download exe and drag&drop your videos on the mp4_merge exe)

We're working with DJI to provide a better way to automatically detect the files

AdrianEddy, 2022-08-30:
In latest dev build, you can simply drag&drop all parts of the video into Gyroflow, and it will ask if you want to merge them
https://gyroflow.xyz/devbuild/?autodownload
[image]

closed 2022-08-30 by AdrianEddy