gyroflow/gyroflow #449: Gyroflow fcpx plugin

davebeech, 2022-08-30:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Is it possible to create a fcpx plugin for gyro data stabilization?
It is really difficult with the workflow.
When i use Gyro I want to color grade and cut after stabilizing, but the rendered video from gyroflow is in worse quality than the original and many informations from the colors get lost.

AdrianEddy, 2022-08-30:
It may be added in the future, but not for now.
What issues with quality are you encountering? It was thoroughly tested and if you use codec like ProRes, the quality is definitely not lost. If you're seeing darker image, uncheck `Advanced->Use GPU decoding` before rendering, that's one known bug in Apple decoders.

davebeech, 2022-08-30:
> 

I actually had those issues. Thank you so much! I'll be totally blessed as soon as all the editing softwares will have gyro stabilization implemented. But with the new gyro stabilization in Resolve for the Black Magic Cameras my hopes are high.

closed 2022-08-30 by davebeech