gyroflow/gyroflow #450: Empty Timeline

Ricboy5, 2022-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.2.0

### What operating system are you using?

Windows 10

### What GPU are you using?

NVIDIA GeForce GTX 630

### What happened?

When I import my clip -shot with a Sony a7 iii- the timeline remains empty. Also the Synchronization doesn`t work

AdrianEddy, 2022-09-01:
Please send me that clip. Not every mode in camera records gyro data though, this is probably normal

AdrianEddy, 2022-09-01:
WeTransfer, google drive, Dropbox  etc

Ricboy5, 2022-09-01:
sent