gyroflow/gyroflow #452: Lens Auto calibration cause the app to crash in MacOS Monterey

estebanavv, 2022-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.2

### What operating system are you using?

Monterey 12.5.1

### What GPU are you using?

AMD RX580

### What happened?

In Monterey I am able to load the video for Lens Calibration but when I click on AutoCalibration it crashes. 

### Relevant log output

```shell
Error reported by Apple Diagnostics (Problem report)

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               gyroflow [40603]
Path:                  /Applications/Gyroflow.app/Contents/MacOS/gyroflow
Identifier:            xyz.gyroflow
Version:               1.0.0 (1)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-09-02 12:33:41.0284 -0400
OS Version:            macOS 12.5.1 (21G83)
Report Version:        12
Anonymous UUID:        25A33A02-7E9F-4F31-7814-57D89428CFB9

Sleep/Wake UUID:       AC57D1C2-B964-4413-90F6-3912A8F5DCD6

Time Awake Since Boot: 100000 seconds
Time Since Wake:       11952 seconds

System Integrity Protection: enabled

Crashed Thread:        0  main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   gyroflow [40603]

Thread 0 Crashed:: main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80e7e997a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80e7e9ce8 mach_msg + 56
2   CoreFoundation                	    0x7ff80e8ed39d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff80e8eba28 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff80e8eae6c CFRunLoopRunSpecific + 562
5   HIToolbox                     	    0x7ff8175995e6 RunCurrentEventLoopInMode + 292
6   HIToolbox                     	    0x7ff81759934a ReceiveNextEventCommon + 594
[... 246 lines cut ...]
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 850 EVO 500GB, 500.11 GB
Serial ATA Device: SHGS31-500GS-2, 500.11 GB
Serial ATA Device: ST31000528AS, 1 TB
USB Device: USB30Bus
USB Device: hub_device
USB Device: AmazonBasics Hard Drive Enclos
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB3.0 Card Reader
USB Device: hub_device
USB Device: USB 2.0 Camera
USB Device: Keyboard Hub
USB Device: USB Optical Mouse
USB Device: Apple Keyboard
USB Device: composite_device
USB Device: USB31Bus
USB Device: AppleUSBUserHCI
USB Device: AppleUSBUserHCI
Thunderbolt Bus: 
```
```

AdrianEddy, 2022-09-02:
Does the same file work in normal mode? Can you also attach a terminal output when running `./gyroflow` directly from terminal from `Show package contents->Contents->MacOS`?

estebanavv, 2022-09-22:
After rebooting te machine I have not been able to reproduce it.

closed 2022-09-22 by estebanavv