gyroflow/gyroflow #453: Crash on launch on Linux with "memory allocation of 48 bytes failed"

markfickett, 2022-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.2.0

### What operating system are you using?

Ubuntu 20.04.3 LTS focal / Linux 5.4.0-86-generic #97-Ubuntu x86_64

### What GPU are you using?

Intel Corporation HD Graphics 620 (rev 02)

### What happened?

GyroFlow crashes on launch with "memory allocation failed".

I should have 12GB or RAM available:
```
free -h
              total        used        free      shared  buff/cache   available
Mem:           15Gi       2.3Gi       6.8Gi       328Mi       6.2Gi        12Gi
Swap:         2.0Gi          0B       2.0Gi
```

### Relevant log output

```shell
./Gyroflow-linux64.AppImage
MDK 0.15.0 (git b9a7944) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 17:35:57 Aug 25 2022
mdk plugin 'mdk-braw': 0x560418116db0
18:47:45 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:47:45 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:47:45 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:47:45 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:47:45 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:47:45 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:47:45 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:47:45 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

18:47:51 [INFO] Latest version: v1.2.0, current version: 1.2.0 (gh709)
18:47:52 [INFO] Lens profiles directory: "/tmp/.mount_Gyrofl9FMz3x/camera_presets"
18:47:52 [INFO] Loaded 3833 lens profiles in 222.340ms
18:47:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) Gen9 HD Graphics NEO"]
18:47:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel Gen OCL Driver") ["Intel(R) HD Graphics Kabylake ULT GT2"]
18:47:52 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) Gen9 HD Graphics NEO
18:47:52 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) corporation intel(r) gen9 hd graphics neo
memory allocation of 48 bytes failed
Aborted (core dumped)
```

AdrianEddy, 2022-09-02:
Does the same happen in .tar.gz package?

markfickett, 2022-09-02:
I wasn't able to run the `.tar.gz`, since it couldn't find `libc++.so.1`.

```
$ ./gyroflow
./gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory
```

I am able to run Gyroflow on the same hardware when booted into Windows, but I do most of my work in Linux so it would be handy to be able to run without rebooting.

Thanks for taking a look!

AdrianEddy, 2022-09-02:
> I wasn't able to run the .tar.gz, since it couldn't find libc++.so.1.

You need to install `libc++-dev`, and probably other needed packages listed in README: 

- Make sure you have latest graphics drivers installed
- Possibly needed packages: `sudo apt install libva2 libvdpau1 libasound2 libxkbcommon0 libpulse0 libc++-dev vdpau-va-driver libvulkan1`
- GPU specific packages:
    - NVIDIA: `nvidia-opencl-icd nvidia-vdpau-driver nvidia-egl-icd nvidia-vulkan-icd libnvcuvid1 libnvidia-encode1`
    - Intel: `intel-media-va-driver i965-va-driver beignet-opencl-icd intel-opencl-icd`
    - AMD: `mesa-vdpau-drivers mesa-va-drivers mesa-opencl-icd libegl-mesa0 mesa-vulkan-drivers`

markfickett, 2022-09-03:
I had given up on that after getting some error before without paying much attention, when the AppImage worked better. Now I've gone through it and am able to install the base packages and the GPU/Intel packages, except for:

```
$ sudo apt-get install vdpau-va-driver
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package vdpau-va-driver is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  mesa-va-drivers mesa-va-drivers:i386

E: Package 'vdpau-va-driver' has no installation candidate
```

I installed `mesa-va-drivers` and `mesa-va-drivers:i386`.

Having done that, the output from the `.tar.gz` looks about the same as the `AppImage`, here's what I get when I run the `.tar.gz`:

```
$ ./gyroflow
MDK 0.15.0 (git b9a7944) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 17:35:57 Aug 25 2022
mdk plugin 'mdk-braw': 0x55b8840864f0
00:43:22 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
00:43:22 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
00:43:22 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
00:43:22 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
00:43:22 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
00:43:22 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
00:43:22 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
00:43:22 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
00:43:26 [INFO] Latest version: v1.2.0, current version: 1.2.0 (gh709)
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

00:43:28 [INFO] Lens profiles directory: "/path/to/.../Gyroflow/camera_presets"
00:43:28 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) Gen9 HD Graphics NEO"]
00:43:28 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel Gen OCL Driver") ["Intel(R) HD Graphics Kabylake ULT GT2"]
00:43:28 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) Gen9 HD Graphics NEO
00:43:28 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) corporation intel(r) gen9 hd graphics neo
00:43:28 [INFO] Loaded 3833 lens profiles in 299.690ms
memory allocation of 48 bytes failed
Aborted (core dumped)
```

AdrianEddy, 2022-09-03:
Can you run it under `gdb`?
```shell
gdb ./gyroflow
```
then type `r` and hit enter
and after it crashes type `bt`, enter and let me know the output

markfickett, 2022-09-03:
```
$ gdb gyroflow
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gyroflow...
(No debugging symbols found in gyroflow)
(gdb) r
Starting program: /.../gyroflow/Gyroflow/gyroflow
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
MDK 0.15.0 (git b9a7944) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 17:35:57 Aug 25 2022
[New Thread 0x7fffefe38700 (LWP 46042)]
[New Thread 0x7fffe7fff700 (LWP 46043)]
[New Thread 0x7fffefc37700 (LWP 46044)]
[New Thread 0x7fffefa36700 (LWP 46045)]
Dwarf Error: DW_FORM_strx1 found in non-DWO CU [in module /.../gyroflow/Gyroflow/lib/libmdk-braw.so]
mdk plugin 'mdk-braw': 0x555556f0f4f0
[New Thread 0x7fffeed54700 (LWP 46046)]
[New Thread 0x7fffee33c700 (LWP 46047)]
17:15:47 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:15:47 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:15:47 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:15:47 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:15:47 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
17:15:47 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
17:15:47 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
17:15:47 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
[New Thread 0x7fffe64eb700 (LWP 46050)]
[New Thread 0x7fffe5cea700 (LWP 46051)]
[New Thread 0x7fffe54e9700 (LWP 46052)]
[New Thread 0x7fffe4ce8700 (LWP 46053)]
[New Thread 0x7fffca849700 (LWP 46054)]
[New Thread 0x7fffca048700 (LWP 46079)]
[New Thread 0x7fffc9847700 (LWP 46080)]
[New Thread 0x7fffc9046700 (LWP 46081)]
[New Thread 0x7fffc8845700 (LWP 46082)]
[New Thread 0x7fffa0da8700 (LWP 46083)]
17:15:53 [INFO] Latest version: v1.2.0, current version: 1.2.0 (gh709)
[New Thread 0x7fff93295700 (LWP 46086)]
[Thread 0x7fff93295700 (LWP 46086) exited]
[New Thread 0x7fff93295700 (LWP 46090)]
[New Thread 0x7fff9141b700 (LWP 46091)]
[New Thread 0x7fff90c1a700 (LWP 46092)]
[New Thread 0x7fff83fff700 (LWP 46093)]
[New Thread 0x7fff837fe700 (LWP 46094)]
[New Thread 0x7fff82ffd700 (LWP 46095)]
[New Thread 0x7fff827fc700 (LWP 46096)]
[New Thread 0x7fff81ffb700 (LWP 46097)]
[New Thread 0x7fff817fa700 (LWP 46098)]
[New Thread 0x7fff80ff9700 (LWP 46099)]
[New Thread 0x7fff63fff700 (LWP 46100)]
[New Thread 0x7fff637fe700 (LWP 46101)]
[New Thread 0x7fff62ffd700 (LWP 46102)]
[New Thread 0x7fff627fc700 (LWP 46103)]
[New Thread 0x7fff61ffb700 (LWP 46104)]
[New Thread 0x7fff617fa700 (LWP 46105)]
[... 28 lines cut ...]
17:15:54 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) Gen9 HD Graphics NEO"]
17:15:54 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel Gen OCL Driver") ["Intel(R) HD Graphics Kabylake ULT GT2"]
17:15:54 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) Gen9 HD Graphics NEO
17:15:54 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) corporation intel(r) gen9 hd graphics neo
[New Thread 0x7fff83fff700 (LWP 46112)]

Thread 39 "QSGRenderThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff83fff700 (LWP 46112)]
0x00007fffe72cab24 in ?? () from /usr/lib/x86_64-linux-gnu/dri/iris_dri.so
(gdb) bt
#0  0x00007fffe72cab24 in ?? () from /usr/lib/x86_64-linux-gnu/dri/iris_dri.so
#1  0x00007fffe7484ebe in ?? () from /usr/lib/x86_64-linux-gnu/dri/iris_dri.so
#2  0x00007fffe72a54c7 in ?? () from /usr/lib/x86_64-linux-gnu/dri/iris_dri.so
#3  0x00007fffe72a6af6 in ?? () from /usr/lib/x86_64-linux-gnu/dri/iris_dri.so
#4  0x00007fffe66bbb26 in ?? () from /usr/lib/x86_64-linux-gnu/dri/iris_dri.so
#5  0x00007ffff4f4819f in ?? () from /.../gyroflow/Gyroflow/lib/libQt6Gui.so.6
#6  0x00007ffff4f4964f in ?? () from /.../gyroflow/Gyroflow/lib/libQt6Gui.so.6
#7  0x00007ffff4de4f38 in QRhi::endFrame(QRhiSwapChain*, QFlags<QRhi::EndFrameFlag>) () from /.../gyroflow/Gyroflow/lib/libQt6Gui.so.6
#8  0x00007ffff47e5ef0 in ?? () from /.../gyroflow/Gyroflow/lib/libQt6Quick.so.6
#9  0x00007ffff47e7167 in ?? () from /.../gyroflow/Gyroflow/lib/libQt6Quick.so.6
#10 0x00007ffff56638ef in ?? () from /.../gyroflow/Gyroflow/lib/libQt6Core.so.6
#11 0x00007ffff3e4a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#12 0x00007ffff3c20293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)
```

markfickett, 2022-09-03:
I made an attempt at building from source.

I got some errors about missing libraries, and needed to `sudo apt-get install libavutil-dev libavcodec-dev libavformat-dev libavfilter-dev libclang-dev`.

I am stuck with `error: header '/usr/include/libswscale/swscale.h' does not exist.` Trying `sudo apt-get install libswscale-dev` but it says it's already the newest version. Looks like the file is at `/usr/include/x86_64-linux-gnu/libswscale/swscale.h`. Not sure if there's just an include path I need to adjust.

AdrianEddy, 2022-09-03:
You need to set correct `FFMPEG_DIR`, like in `__env-linux.sh`
FFmpeg should be 5.1, ideally downloaded  by `install-deps-linux.sh` like in the building steps

AdrianEddy, 2022-09-03:
That gdb stack trace suggests an issue in Intel GPU drivers. you can try setting this:
`export QSG_RHI_BACKEND=vulkan`

markfickett, 2022-09-03:
Setting that does let it launch, hooray! I just had to enable DRI3 support in my Xorg config file and restart (I followed https://donjajo.com/enable-intel-dri3-xorg/ ).
[image]

When I try to load a video (about 30s long at 2.7k), I see a spinner in the UI for more than 1 minute. In the console (for the binary from the tar or from the AppImage) I see:

```
$ QSG_RHI_BACKEND=vulkan ./Gyroflow-linux64.AppImage 
MDK 0.15.0 (git b9a7944) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 17:35:57 Aug 25 2022
mdk plugin 'mdk-braw': 0x555dc204de20
18:58:16 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:58:16 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:58:16 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:58:16 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
18:58:16 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:58:17 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:58:17 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:58:17 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
18:58:19 [INFO] Latest version: v1.2.0, current version: 1.2.0 (gh709)
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

18:58:19 [INFO] Lens profiles directory: "/tmp/.mount_GyroflrJMSQT/camera_presets"
18:58:19 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) Gen9 HD Graphics NEO"]
18:58:19 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel Gen OCL Driver") ["Intel(R) HD Graphics Kabylake ULT GT2"]
18:58:19 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) Gen9 HD Graphics NEO
18:58:19 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) corporation intel(r) gen9 hd graphics neo
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

18:58:19 [INFO] Loaded 3833 lens profiles in 266.944ms
18:58:25 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/markfickett/Documents/markfickett/appbkp/gyroflow/mov/RC5_OR_0000.MP4"
18:58:25 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:58:25 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.video" ""
18:58:25 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
18:58:25 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.audio" "open"
18:58:25 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
18:58:25 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.MMAL.mmal"
18:58:25 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:58:25 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
18:58:25 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
18:58:25 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
18:58:25 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
18:58:25 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
18:58:25 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "decoder.audio" "open"
18:58:25 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
18:58:25 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
18:58:25 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
18:58:26 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
18:58:26 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
```

AdrianEddy, 2022-09-04:
It turned out the vulkan video playback was not compiled, please try in latest dev build https://gyroflow.xyz/devbuild/?autodownload

markfickett, 2022-09-04:
Thanks!

I got this error:

```
11:21:40 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
11:21:40 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.video" ""
11:21:40 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
11:21:40 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
11:21:40 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
11:21:40 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.MMAL.mmal"
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:21:40 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
```
Then I installed that lib with `sudo apt-get install libvdpau-va-gl1`.

Now I'm getting this error:
```
11:24:00 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
11:24:00 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.video" ""
11:24:00 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
11:24:00 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
11:24:00 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
11:24:00 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:24:00 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
11:24:00 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.VDPAU"
11:24:00 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:24:00 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
11:24:01 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:24:01 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
11:24:01 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
11:24:01 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
```

Looking at [this ticket](https://github.com/intel/intel-vaapi-driver/issues/455) for inspiration, here's some more debug info:

```
$ sudo apt-get install vainfo
$ vainfo
libva info: VA-API version 1.7.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.7 (libva 2.6.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.1.1 ()
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :VAEntrypointVLD
      VAProfileMPEG2Main              :VAEntrypointVLD
      VAProfileH264Main               :VAEntrypointVLD
      VAProfileH264Main               :VAEntrypointEncSliceLP
      VAProfileH264High               :VAEntrypointVLD
      VAProfileH264High               :VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :VAEntrypointVLD
      VAProfileJPEGBaseline           :VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :VAEntrypointVLD
      VAProfileHEVCMain               :VAEntrypointVLD
      VAProfileHEVCMain10             :VAEntrypointVLD
      VAProfileVP9Profile0            :VAEntrypointVLD
      VAProfileVP9Profile2            :VAEntrypointVLD
$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep libva
libva-dev: 2.7.0-2
libva-drm2: 2.7.0-2
libva-glx2: 2.7.0-2
libva-wayland2: 2.7.0-2
libva-x11-2: 2.7.0-2
libva2: 2.7.0-2
$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep va-driver
i965-va-driver: 2.4.0-0ubuntu1
intel-media-va-driver: 20.1.1+dfsg1-1
mesa-va-drivers: 21.2.6-0ubuntu0.1~20.04.2
mesa-va-drivers: 21.2.6-0ubuntu0.1~20.04.2
va-driver-all: 2.7.0-2
```

AdrianEddy, 2022-09-04:
hmm, it still says "Vulkan support not compiled", are you sure you used latest dev build? does it say (gh741) in title?

markfickett, 2022-09-04:
Yep, launched from the AppImage in the dev build zip:

[image]

But I see I'm getting the same error about `no function __vaDriverInit_1_0` from the release (gh709) build too. (I also tried the `.tar.gz` from the dev build, and I see the 'not compiled' error as well as the spinner in the UI.)

Here's the full output:

Release build (gh709 in the UI title):

```
$ QSG_RHI_BACKEND=vulkan ./Gyroflow-linux64.AppImage
MDK 0.15.0 (git b9a7944) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 17:35:57 Aug 25 2022
mdk plugin 'mdk-braw': 0x563fc62ecdb0
14:49:46 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:49:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:49:46 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:49:46 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:49:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:49:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:49:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:49:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:49:51 [INFO] Latest version: v1.2.0, current version: 1.2.0 (gh709)
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

14:49:52 [INFO] Lens profiles directory: "/tmp/.mount_Gyroflbdkiy9/camera_presets"
14:49:53 [INFO] Loaded 3833 lens profiles in 209.932ms
14:49:53 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) Gen9 HD Graphics NEO"]
14:49:53 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel Gen OCL Driver") ["Intel(R) HD Graphics Kabylake ULT GT2"]
14:49:53 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) Gen9 HD Graphics NEO
14:49:53 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) corporation intel(r) gen9 hd graphics neo
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

14:50:09 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:50:09 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:50:10 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/markfickett/Documents/markfickett/appbkp/gyroflow/mov/RC5_OR_0000.MP4"
14:50:10 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
14:50:10 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.video" ""
14:50:10 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
14:50:10 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.audio" "open"
14:50:10 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
14:50:10 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.VDPAU"
14:50:10 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
14:50:10 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
14:50:10 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
14:50:10 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
14:50:10 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
14:50:11 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
14:50:11 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "decoder.audio" "open"
14:50:11 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
14:50:11 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
14:50:11 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
14:50:11 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
14:50:11 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
```

And from the dev build (I see it does say built Sep 3):

```
$ QSG_RHI_BACKEND=vulkan ./Gyroflow-linux64.AppImage
MDK 0.16.0 (git b724cce) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 15:52:39 Sep  3 2022
mdk plugin 'mdk-braw': 0x55cdd279cdc0
[... cut ...]
14:53:06 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
14:53:06 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] Vulkan support not compiled
```

By the way, I really appreciate your debugging the Linux build, and I'm glad to continue to keep working on it too. I did run the Windows version on a test video and it was super easy to do / got the results I was looking for, so if at some point you want to focus efforts elsewhere I'll definitely be able to dual-boot to do what I need.

AdrianEddy, 2022-09-07:
Please try gh749 version https://gyroflow.xyz/devbuild/?autodownload

markfickett, 2022-09-07:
Success!

With that build I'm able to open Gyroflow, load a video, and view/render stabilized output. (The experimental zero-copy GPU preview seems fine too. And I think, using footage from the RunCam 5 Orange, I get the same results whether I just use the gyro data it finds in the video file or if I load the separate .gcsv file.)

I still see some error logs in the console which I'll copy below, but they don't seem to be preventing me from doing anything.

```
gh749$ QSG_RHI_BACKEND=vulkan ./Gyroflow-linux64.AppImage
MDK 0.16.0 (git bc8f174) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.6 GCC4.2.1; libc++14000; 15:06:24 Sep  6 2022
mdk plugin 'mdk-braw': 0x565219833dd0
16:52:19 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:52:19 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:52:19 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:52:19 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:52:19 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
16:52:19 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
16:52:19 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
16:52:19 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
16:52:21 [DEBUG] (4) gyroflow::util: Minidump uploaded: OK
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

16:52:25 [INFO] Latest version: v1.2.0, current version: 1.2.0 (gh749)
16:52:25 [INFO] Lens profiles directory: "/tmp/.mount_Gyrofl1peTq6/camera_presets"
16:52:25 [INFO] Loaded 3808 lens profiles in 195.534ms
16:52:25 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) Gen9 HD Graphics NEO"]
16:52:25 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel Gen OCL Driver") ["Intel(R) HD Graphics Kabylake ULT GT2"]
16:52:25 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) Gen9 HD Graphics NEO
16:52:25 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) corporation intel(r) gen9 hd graphics neo
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

16:52:27 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:52:27 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:52:32 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/markfickett/Documents/markfickett/appbkp/gyroflow/mov/RC5_OR_0001.MP4"
16:52:33 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
16:52:33 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.video" ""
16:52:33 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
16:52:33 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
16:52:33 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
16:52:33 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
16:52:33 [DEBUG] (9) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f4165585660) QSize(1221, 916)
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
16:52:33 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.VDPAU"
16:52:33 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
16:52:33 [INFO] surface size: 960x720
16:52:33 [INFO] surface size: 960x720
16:52:33 [DEBUG] (2) gyroflow_core::camera_identifier: Camera ID Brand: Runcam, Model: RC5-OR, Metadata: Object {"fwversion": String("V2.0.3"), "id": String("RC5-OR"), "note": String("development_test"), "timestamp": String("1644159993"), "vendor": String("Runcam"), "version": String("1.1"), "videofilename": String("RC5_OR_0000.MP4")}
16:52:33 [INFO] Camera identifier string: runcam-rc5-or-wide-1920x1440@59940
16:52:33 [DEBUG] (2) telemetry_parser::util: Reading duration: (None, None, None)
16:52:33 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Vulkan
16:52:33 [DEBUG] (9) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f416508c3e0) QSize(960, 720)
[AVHWFramesContext @ 0x7f41308d7b40] Unsupported sw format: yuvj420p
[h264 @ 0x7f41308f1140] Failed setup for format vdpau: hwaccel initialisation returned error.
[... cut ...]
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
16:52:34 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_VDPAU (vdpau) with format Some(AV_PIX_FMT_VDPAU)
[AVHWFramesContext @ 0x7f41980880c0] Unsupported sw format: yuvj420p
[h264 @ 0x7f41bc12d780] Failed setup for format vdpau: hwaccel initialisation returned error.
16:52:42 [INFO] Setting offset at 6398.0585: -31.3164 (cost 2049.8662)
```

AdrianEddy, 2022-09-07:
I'm glad it's working for you, so the Vulkan build on linux was not compiled properly, that is now fixed.
As for the original crash, there's really not much I can do, it's a crash in a driver, because of something Qt does when rendering the UI, I don't have control over that, so I'm closing this issue

closed 2022-09-07 by AdrianEddy