gyroflow/gyroflow #454: CLI option to create gyroflow projects

EmberLightVFX, 2022-09-04:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi! I'm the developer of [Gyroflow to CSV](https://github.com/EmberLightVFX/GyroFlow-to-CSV).
I'm currently working with Andrew Hazelden to make a BMD Fusion script that makes it super quick and easy to get the stabilize-data into a 3D camera inside of Fusion.
It currently work in tandem with Gyroflow to CSV but you still have to open up gyroflow and create the file each time.

What would make the workflow even more awesome would be if the script automatically could create the gyroflow projct directly from within Fusion!
The current CLI options that were released in beta 3 days ago seem to only allow rendering of the footage.
To make our idea of the script work we would need to analyze a video file, set the stabilization to fixed camera and save the project including processed gyro data to specific location.

A BIG bonus would be if it also could auto-sync the footage before saving the project so we would get even better result.

In the end, if it would be possible to set all UI options in Gyroflow through CLI (framerate, rotation, IMU orientation, integration method, max sync points, flow method and so on), Gyroflow could easily be integrated into a lot of DCC softwares and pipelines everywhere! But one step at a time :)

AdrianEddy, 2022-09-04:
> In the end, if it would be possible to set all UI options in Gyroflow through CLI (framerate, rotation, IMU orientation, integration method, max sync points, flow method and so on), Gyroflow could easily be integrated into a lot of DCC softwares and pipelines everywhere! But one step at a time :)

It is already possible to do that, that's what presets are for. You can load the preset from CLI together with videos, and you can set all settings there, do the autosync etc. I'll add exporting project file instead of rendering soon

AdrianEddy, 2022-09-04:
added in 761e73495c9f1ca4c42cfb873e56228164112dbf, please try latest dev build once it completes building https://github.com/gyroflow/gyroflow/actions/runs/2989350360

AndrewHazelden, 2022-09-05:
@AdrianEddy Thanks for your assistance on this feature request. I downloaded the Gyroflow dev build last night and examined the CLI parameters that are now available. Excellent work!

I will report back in more detail as soon as I finish running several video clips through the CLI to better understand the whole process.

AdrianEddy, 2022-09-07:
I tested it again and it works as expected for me, if you have any other issues or feature requests, let me know.

closed 2022-09-07 by AdrianEddy