gyroflow/gyroflow #464: Gyroflow v1.2.0 not opening on Windows 11 / RTX3080

underlines, 2022-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.2.0

### What operating system are you using?

Windows 11 Pro x64 EN_INT

### What GPU are you using?

MSI RTX3080

### What happened?

1. Installed latest Nvidia driver 516.94
2. Extracted Gyroflow v1.2.0 to D:\ml\gyroflow\
3. Run gyroflow.exe
4. Observing the process running in taskmgr, but no window appearing, logging 2 warnings for **QQmlApplicationEngine failed to load** and **qrc:/src/ui/main_window.qml: No such file or directory**
5. Killing the process
6. Using the following batch to try again:
```
@echo off
set NO_OPENCL=1
set RUST_BACKTRACE=full
Gyroflow.exe
pause
```
Same issue

### Relevant log output

```shell
19:02:53 [INFO] [MDK] MDK 0.15.0 (git e8c668a) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1933; MSVCRT14.33.31629.0/msvcp140.dll; 15:32:35 Aug 23 2022
19:02:53 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
19:02:53 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
19:02:53 [DEBUG] (1) mdk: [MDK] user module: C:\Users\Jan\Downloads\Gyroflow-windows64\Gyroflow.exe
19:02:53 [DEBUG] (1) mdk: [MDK] user module: C:\Users\Jan\Downloads\Gyroflow-windows64\Gyroflow.exe
19:02:53 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
19:02:53 [WARN] QQmlApplicationEngine failed to load component
19:02:53 [WARN] qrc:/src/ui/main_window.qml: No such file or directory
19:02:53 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "NVIDIA GeForce RTX 3080", vendor: 4318, device: 8726, device_type: DiscreteGpu, backend: Vulkan }
19:02:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
19:02:54 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
19:02:54 [DEBUG] (1) gyroflow::rendering: GPU type: NVIDIA, from name: nvidia geforce rtx 3080
```

AdrianEddy, 2022-09-16:
does the same happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload
You can try the workaround from #412

underlines, 2022-09-17:
> does the same happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload You can try the workaround from #412

Definitely the same root cause as #412 
Solved by manually copying the /src/ui/ folder to root / where gyroflow.exe lies.

AdrianEddy, 2022-09-17:
Did you check the latest dev build first? I can't reproduce this issue on any of my devices, but the Qt library was updated in latest dev build, so I'm curious if that helped

AdrianEddy, 2022-09-30:
Could you please test latest dev build? Qt was updated to new version so I'm curious if that fixed it https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-10-17:
I believe this is fixed in v1.3.0

closed 2022-10-17 by AdrianEddy