gyroflow/gyroflow #468: Gyroflow v1.2.0 not opening on Windows 10 / GTX1050

antokubus, 2022-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

Gyroflow-windows64 - 120

### What operating system are you using?

windows 10

### What GPU are you using?

Windows 10 Enterprise

### What happened?

need help.
Gyroflow v1.2.0 not opening on my Windows 10 / GTX 1050
its always force close.

already try the other version, but always force close.

### Relevant log output

```shell
01:43:36 [INFO] [MDK] MDK 0.15.0 (git e8c668a) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1933; MSVCRT14.33.31629.0/msvcp140.dll; 15:32:35 Aug 23 2022
01:43:36 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
01:43:36 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
01:43:36 [DEBUG] (1) mdk: [MDK] user module: D:\2_garage\Gyroflow-windows64 120\Gyroflow.exe
01:43:36 [DEBUG] (1) mdk: [MDK] user module: D:\2_garage\Gyroflow-windows64 120\Gyroflow.exe
01:43:36 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
01:43:41 [INFO] Latest version: v1.2.0, current version: 1.2.0
01:43:41 [INFO] Lens profiles directory: "\\\\?\\D:\\2_garage\\Gyroflow-windows64 120\\camera_presets"
01:43:41 [INFO] Loaded 3833 lens profiles in 486.951ms
```

AdrianEddy, 2022-09-20:
Do you have latest NVIDIA drivers and latest windows updates installed?

antokubus, 2022-09-20:
before i write in here, i already do that.
but still cant work :-(

antokubus, 2022-09-20:
[image]

i try to download the other version.
gyroflow ver 100 rc 1 is the only one that can be open.
but, it cant works normally

AdrianEddy, 2022-09-20:
does v1.0.0-rc5 work?

antokubus, 2022-09-20:
nop, the only one, v1.0.0-rc1 
but, when load the file. its force close again

[image]

AdrianEddy, 2022-09-20:
Can you run `Gyroflow_with_console.bat` from latest dev build https://gyroflow.xyz/devbuild/?autodownload  and show me a screenshot of the console after it closes?

antokubus, 2022-09-20:
[image]

do yo mean this ?

or this :
@echo off
set RUST_BACKTRACE=full
Gyroflow.exe
pause

ChoiHalim75, 2022-09-24:
I have the same issue.
I just updated the graphics card and nvidia drive, 1.0 rc1 runs, but all other versions are closed after a crash.
All versions worked fine before the upgrade.
The new graphics card is the 3060.

antokubus, 2022-09-24:
> Can you run `Gyroflow_with_console.bat` from latest dev build https://gyroflow.xyz/devbuild/?autodownload and show me a screenshot of the console after it closes?

need help :-(

AdrianEddy, 2022-09-24:
Are you sure you installed latest Nvidia drivers from nvidia.com? The ones included with windows are older

ChoiHalim75, 2022-09-24:
> AdrianEddy

This is a console screenshot tested with the last build of your link.

Of course, this is the latest driver from the NVIDIA official [website.]([image])

ChoiHalim75, 2022-09-24:
[image]

antokubus, 2022-09-25:
> Are you sure you installed latest Nvidia drivers from nvidia.com? The ones included with windows are older

problem solved for me.
my nvidia is already the last version.
but, everything works well when i reinstall the intel graphic driver = gfx_win_101.3413_101.2111

thanks @AdrianEddy :-)

[image]

ChoiHalim75, 2022-09-25:
> Are you sure you installed latest Nvidia drivers from nvidia.com? The ones included with windows are older

After setting 'set NO_OPENCL=1' in the bat file,
When I run bat, the window opens normally.
As a result of simple processing with the sample file I took, the rest seem to be fine.
I am always grateful for the great program and kind response.

AdrianEddy, 2022-09-30:
Could you please test latest dev build? Qt was updated to new version so I'm curious if that fixed it https://gyroflow.xyz/devbuild/?autodownload

ChoiHalim75, 2022-09-30:
> Could you please test latest dev build? Qt was updated to new version so I'm curious if that fixed it https://gyroflow.xyz/devbuild/?autodownload

[image]

AdrianEddy, 2022-11-18:
Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-11-21:
I believe this is either fixed in latest dev build or a driver issue, please make sure you have all newest drivers installed and all windows updates, and if it still happens, please reopen
 https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-21 by AdrianEddy

antokubus, 2022-11-21:
Thank you 🙏for your response