gyroflow/gyroflow #472: GPU-Encoding Bitrate does not match

xviews, 2022-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.2.0 / lastDevBuild

### What operating system are you using?

Windows 10 21H2

### What GPU are you using?

AMD 6900XT

### What happened?

If GPU encoding is used, then the bitrate does not match the defined bitrate.
If only CPU encoding is used, then the bitrate match the defined bitrate.
No matter which bitrate I set in Gyroflow, it stays at 20 Mbits in h264 and h265

MediaInfo
Format/String                    : MPEG-4
Format_Profile                   : Base Media
CodecID/String                   : isom (isom/iso2/mp41)
FileSize/String                  : 129 MiB
Duration/String                  : 54s 105 ms
OverallBitRate/String            : 20,0 Mb/s
Encoded_Date                     : UTC 2022-10-05 09:21:01
Tagged_Date                      : UTC 2022-10-05 09:21:01
Encoded_Application/String       : Lavf59.27.100
Comment                          : EIS:OFF,FOV:Normal

[image]
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9714470/gyroflow.log)

### Relevant log output

```shell
08:43:27 [DEBUG] (1) mdk: [MDK] default000001ACE055C4F0 new FrameReader...
08:43:27 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound null
08:43:27 [DEBUG] (1) mdk: [MDK] XAudio2_9.dll
08:43:27 [DEBUG] (1) mdk: [MDK] Build with XAudio2 from Win8+ SDK
08:43:27 [DEBUG] (1) mdk: [MDK] Try symbol 'XAudio2Create' from Windows SDK dll
08:43:27 [DEBUG] (1) mdk: [MDK] default000001ACE055C060 new FrameReader...
08:43:27 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound null
08:43:27 [DEBUG] (1) mdk: [MDK] 000001ACE4899610__cdecl mdk::abi::Player::Player(void)
08:43:27 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 ignore unsupported decoder BRAW:gpu=auto:scale=1920x1080
08:43:27 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C060 ignore unsupported decoder BRAW:gpu=auto:scale=1920x1080
08:43:27 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "C:/Users/USER/Desktop/DJI_0046.MP4"
08:43:27 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 state: 0=>0=>0, 0
08:43:27 [INFO] [MDK] default000001ACE055C4F0 void __cdecl mdk::abi::FrameReader::stop(void), current state: 0
08:43:27 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 state: 0=>0=>0, 0
08:43:27 [INFO] [MDK] default000001ACE055C4F0 void __cdecl mdk::abi::FrameReader::stop(void), current state: 0
08:43:27 [INFO] [MDK] 000001AC9CDF7930 MediaControl.prepare(0, ...) C:/Users/USER/Desktop/DJI_0046.MP4
08:43:27 [DEBUG] (1) mdk: [MDK] default 000001ACE055C4F0 FrameReader.start(0, ...)
08:43:27 [DEBUG] (1) mdk: [MDK] default 000001ACE055C4F0 FrameReader::update MediaStatus 0=>0X2
08:43:27 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 state: 0=>0=>1, 0
08:43:27 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 state: 0=>1=>1, 0
08:43:27 [DEBUG] (26) mdk: [MDK] start frame reader thread: 10088
08:43:27 [DEBUG] (26) mdk: [MDK] try to load av module: C:\Users\USER\Downloads\Gyroflow-windows64\ffmpeg-5.dll
08:43:27 [DEBUG] (26) mdk: [MDK] try to load av module: C:\Users\USER\Downloads\Gyroflow-windows64\avformat-59.dll
08:43:27 [DEBUG] (26) mdk: [MDK] Selected avformat runtime version: 59.27.100 (build: 59.30.100), license: GPL version 2 or later
08:43:27 [DEBUG] (26) mdk: [MDK] Selected avformat runtime configuration: --extra-version=avbuild --disable-doc --disable-debug --enable-shared --enable-runtime-cpudetect --enable-libmfx --disable-ptx-compression --enable-mediafoundation --disable-vulkan --disable-postproc --toolchain=msvc --windres=llvm-windres --extra-windresflags='--target=x86_64-pc-windows-msvc -I/c/Program\ Files/Microsoft\ Visual\ Studio/2022/Enterprise/VC/Tools/MSVC/14.32.31326/ATLMFC/include -I/c/Program\ Files/Microsoft\ Visual\ Studio/2022/Enterprise/VC/Tools/MS
08:43:27 [DEBUG] (26) mdk: [MDK] VC/14.32.31326/include -I/c/Program\ Files\ \(x86\)/Windows\ Kits/NETFXSDK/4.8/include/um -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22621.0/ucrt -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22621.0/shared -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22621.0/um -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22621.0/winrt -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22621.0/cppwinrt' --enable-pic --extra-cflags='-Zi -FS -MD -guard:cf -I/d
[... 211 lines cut ...]
08:43:29 [DEBUG] (27) mdk: [MDK] video stream#0 sending 1 invalid AOT frame @0.000000s. seeking: 0
08:43:29 [DEBUG] (27) mdk: [MDK] VideoRenderer clear buffered frames
08:43:29 [DEBUG] (27) mdk: [MDK] 0-track seek end video frame @0 seek_pos_: -1
08:43:29 [DEBUG] (26) mdk: [MDK] default FrameReader000001ACE055C4F0 #3 seekComplete 0
08:43:29 [DEBUG] (26) mdk: [MDK] default 000001ACE055C4F0 FrameReader::update MediaStatus 0X124=>0X124
08:43:29 [DEBUG] (26) mdk: [MDK] default FrameReader 000001ACE055C4F0 callbacks 1, seeking: 0, seekComplete#3 @0
08:43:32 [DEBUG] (1) mdk: [MDK] 000001AC9CDF7930 void __cdecl mdk::abi::MediaControlPush::setState(enum mdk::abi::State)@707 requested state 2=>2, current state 2. status: 0X124
08:43:32 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 request to pause 1, loaded: 4.
08:43:32 [DEBUG] (1) mdk: [MDK] default FrameReader000001ACE055C4F0 state: 2=>2=>2, 0
08:43:33 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
08:43:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
08:43:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)
08:43:33 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
08:43:33 [DEBUG] (2) gyroflow::rendering: Possible encoders with AMD: [("hevc_amf", true), ("hevc_mf", true), ("libx265", false)]
08:43:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_D3D11VA 0, for: hevc_amf
08:43:33 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_amf"), processing_order: PreConversion
08:43:33 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 54104, render_frame_count: 3243
08:43:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [NV12, YUV420P, D3D11, DXVA2_VLD]
08:43:34 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
08:43:34 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_D3D11VA), encoder_pixel_format: Some(NV12)
08:43:34 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: MPEG
08:43:34 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_D3D11VA)
08:43:34 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x1acefb668c0
08:45:04 [DEBUG] (28) mdk: [MDK] __cdecl ThreadLocal<bool>::Data::~Data(void) thread: 14420
```

xviews, 2022-10-06:
Okay, if I stay under 100 mbits then it fits. 
https://github.com/GPUOpen-LibrariesAndSDKs/AMF/issues/357

AdrianEddy, 2022-11-18:
Can you try adding `-profile high` or `-rc cbr` in the Encoder parameters (Export settings -> Advanced)? let me know which one worked, I'll add it by default

xviews, 2022-11-19:
### **original source file**
[image]

### **-profile high**
[image]

### **-rc cbr**
[image]

### **-rc cqp**
[image]

AdrianEddy, 2022-11-19:
`-rc cqp` ignores the bitrate set in gyroflow, right?
how about `-profile_tier high`?

Here are all options for HEVC AMF encoder, for reference:
```
hevc_amf AVOptions:
  -usage             <int>        E..V....... Set the encoding usage (from 0 to 3) (default transcoding)
     transcoding     0            E..V.......
     ultralowlatency 1            E..V.......
     lowlatency      2            E..V.......
     webcam          3            E..V.......
  -profile           <int>        E..V....... Set the profile (default main) (from 1 to 1) (default main)
     main            1            E..V.......
  -profile_tier      <int>        E..V....... Set the profile tier (default main) (from 0 to 1) (default main)
     main            0            E..V.......
     high            1            E..V.......
  -level             <int>        E..V....... Set the encoding level (default auto) (from 0 to 186) (default auto)
     auto            0            E..V.......
     1.0             30           E..V.......
     2.0             60           E..V.......
     2.1             63           E..V.......
     3.0             90           E..V.......
     3.1             93           E..V.......
     4.0             120          E..V.......
     4.1             123          E..V.......
     5.0             150          E..V.......
     5.1             153          E..V.......
     5.2             156          E..V.......
     6.0             180          E..V.......
     6.1             183          E..V.......
     6.2             186          E..V.......
  -quality           <int>        E..V....... Set the encoding quality (from 0 to 10) (default speed)
     balanced        5            E..V.......
     speed           10           E..V.......
     quality         0            E..V.......
  -rc                <int>        E..V....... Set the rate control mode (from -1 to 3) (default -1)
     cqp             0            E..V....... Constant Quantization Parameter
     cbr             3            E..V....... Constant Bitrate
     vbr_peak        2            E..V....... Peak Contrained Variable Bitrate
     vbr_latency     1            E..V....... Latency Constrained Variable Bitrate
  -header_insertion_mode <int>        E..V....... Set header insertion mode (from 0 to 2) (default none)
     none            0            E..V.......
     gop             1            E..V.......
     idr             2            E..V.......
  -gops_per_idr      <int>        E..V....... GOPs per IDR 0-no IDR will be inserted (from 0 to INT_MAX) (default 1)
  -preanalysis       <boolean>    E..V....... Enable preanalysis (default false)
  -vbaq              <boolean>    E..V....... Enable VBAQ (default false)
  -enforce_hrd       <boolean>    E..V....... Enforce HRD (default false)
  -filler_data       <boolean>    E..V....... Filler Data Enable (default false)
  -max_au_size       <int>        E..V....... Maximum Access Unit Size for rate control (in bits) (from 0 to INT_MAX) (default 0)
  -min_qp_i          <int>        E..V....... min quantization parameter for I-frame (from -1 to 51) (default -1)
  -max_qp_i          <int>        E..V....... max quantization parameter for I-frame (from -1 to 51) (default -1)
  -min_qp_p          <int>        E..V....... min quantization parameter for P-frame (from -1 to 51) (default -1)
  -max_qp_p          <int>        E..V....... max quantization parameter for P-frame (from -1 to 51) (default -1)
  -qp_p              <int>        E..V....... quantization parameter for P-frame (from -1 to 51) (default -1)
  -qp_i              <int>        E..V....... quantization parameter for I-frame (from -1 to 51) (default -1)
  -skip_frame        <boolean>    E..V....... Rate Control Based Frame Skip (default false)
  -me_half_pel       <boolean>    E..V....... Enable ME Half Pixel (default true)
  -me_quarter_pel    <boolean>    E..V....... Enable ME Quarter Pixel  (default true)
  -aud               <boolean>    E..V....... Inserts AU Delimiter NAL unit (default false)
  -log_to_dbg        <boolean>    E..V....... Enable AMF logging to debug output (default false)
  ```
  
  and here for H264/AVC
  ```
  h264_amf AVOptions:
  -usage             <int>        E..V....... Encoder Usage (from 0 to 3) (default transcoding)
     transcoding     0            E..V....... Generic Transcoding
     ultralowlatency 1            E..V.......
     lowlatency      2            E..V.......
[... 34 lines cut ...]
     vbr_peak        2            E..V....... Peak Contrained Variable Bitrate
     vbr_latency     3            E..V....... Latency Constrained Variable Bitrate
  -enforce_hrd       <boolean>    E..V....... Enforce HRD (default false)
  -filler_data       <boolean>    E..V....... Filler Data Enable (default false)
  -vbaq              <boolean>    E..V....... Enable VBAQ (default false)
  -frame_skipping    <boolean>    E..V....... Rate Control Based Frame Skip (default false)
  -qp_i              <int>        E..V....... Quantization Parameter for I-Frame (from -1 to 51) (default -1)
  -qp_p              <int>        E..V....... Quantization Parameter for P-Frame (from -1 to 51) (default -1)
  -qp_b              <int>        E..V....... Quantization Parameter for B-Frame (from -1 to 51) (default -1)
  -preanalysis       <boolean>    E..V....... Pre-Analysis Mode (default false)
[... cut ...]
     cavlc           2            E..V....... Context Adaptive Variable-Length Coding
     cabac           1            E..V....... Context Adaptive Binary Arithmetic Coding
  -me_half_pel       <boolean>    E..V....... Enable ME Half Pixel (default true)
  -me_quarter_pel    <boolean>    E..V....... Enable ME Quarter Pixel (default true)
  -aud               <boolean>    E..V....... Inserts AU Delimiter NAL unit (default false)
  -log_to_dbg        <boolean>    E..V....... Enable AMF logging to debug output (default false)
```

xviews, 2022-11-19:
> -rc cqp ignores the bitrate set in gyroflow, right?

Yes, I need to complete the command >>> -rc cqp -qp_i 28 -qp_p 28 works and I get 130mbits

> -profile_tier high

no change, again 21306 kbits

AdrianEddy, 2022-11-21:
I tested myself on an AMD card and I was able to set bitrate > 100 Mbps without any additional flags, so I guess this bug is only affecting some cards/drivers. 

I now added a message about this behavior so everyone can decide if they want to lower the bitrate or use `-rc cqp` or other options

[image]

closed 2022-11-21 by AdrianEddy

prafgup, 2024-08-04:
I have the same exact issue with my Apple M1 Pro, if I disable GPU encoding it works fine and respects my Bitrate that I have set, but gives out a random Bitrate in case its enabled, I have tried setting the `-rc cqp` custom encoder as well.

cc @AdrianEddy 
[image]

With GPU - 
[image]

With CPU - 
[image]

NikitaKA, 2024-09-19:
`-rc cbr_hq` helped me