gyroflow/gyroflow #476: Left half only rendered

Schweizer33, 2022-10-09:
Insta360 GO2 Pro input file.  2880x2880 lens.  Stabilizes fine in Gyroflow v1.3 but the rendered output 2560x1440 only renders the left half, the right side is blank.  Worse in lower res renders.  Same file and lens in v1.2 work fine.  
WIN64

AdrianEddy, 2022-10-09:
Can you please upload gyroflow.log? What system are you using? What's your GPU? Was GPU encoding enabled? I can't reproduce this on Windows.

Schweizer33, 2022-10-09:
AMD Ryzen 5 3400G .  Win10 home. GPU "unchecked".  Log attached.  Thanks. [](url)
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9742341/gyroflow.log)
Sample here: https://youtu.be/Sq3aSGmV5cc

AdrianEddy, 2022-10-09:
Could you please check latest dev build? Should be fixed https://gyroflow.xyz/devbuild/?autodownload

closed 2022-10-09 by AdrianEddy

Schweizer33, 2022-10-10:
Your fix worked. I tried it with GPU on and off.  Seemed faster too though I only ran a small test clip.  Log attached.  Thanks.  You're amazing and appreciated.  TOM in Minnesota.  
Look how wonderfully your work improves the art of FPV cine.  https://youtu.be/DPJhwcB3GTg
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9744738/gyroflow.log)