gyroflow/gyroflow #477: No GPU encoding

linsenpago, 2022-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0 beta 2

### What operating system are you using?

Win11

### What GPU are you using?

Sapphire Radeon RX470

### What happened?

When starting queue rendering no GPU power is used but it is enabled in the settings.
In 1.2 GPU Hardware is used correctly.

AdrianEddy, 2022-10-09:
Can you please upload gyroflow.log after rendering?

linsenpago, 2022-10-09:
can you please tell me where i can find the gyroflow.log file?

AdrianEddy, 2022-10-09:
It's in the same directory as main program, next to Gyroflow.exe

linsenpago, 2022-10-09:
It seems that it is only be produced when I start the rendering via cmd window on windows? How can I do this? I only used it via GUI..

AdrianEddy, 2022-10-09:
no, it's always created if Gyroflow has rights to write in this folder. If you have it on C:/ or C:/Program files, by default Windows denies access. Run Gyroflow as Administrator in this case, or run Gyroflow from other location (like desktop)

mickey1cx, 2022-10-09:
Hello I have the same problem. Found the next thread https://learn.microsoft.com/en-us/answers/questions/321517/when-compile-hlsl-shader-model51-get-error-x3506-u.html.
 I took the d3dcompiler_47.dll file from last VS Code and replaced it in the Gyroflow directory.

AdrianEddy, 2022-10-09:
@mickey1cx hmm, could you show me your gyroflow.log file before and after replacing the dll?

mickey1cx, 2022-10-09:
@AdrianEddy 
placed logs at https://dropmefiles.com/jI8D7
also check win version 1.2.0, have this problem too
my system - win 10 / GeForce GTX 1050 Ti

AdrianEddy, 2022-10-09:
Could you please check latest dev build? Should be fixed @mickey1cx @linsenpago https://gyroflow.xyz/devbuild/?autodownload

mickey1cx, 2022-10-09:
> Could you please check latest dev build? Should be fixed @mickey1cx @linsenpago https://gyroflow.xyz/devbuild/?autodownload

No, have error. Encoding on CPU.

21:43:31 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "NVIDIA GeForce GTX 1050 Ti", vendor: 4318, device: 7298, device_type: DiscreteGpu, backend: Dx12 }
21:43:31 [ERROR] thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_render_pipeline
    Internal error in VERTEX shader: D3DCompile error (0x8876086C): error X3506: unrecognized compiler target 'vs_5_1'

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9742619/gyroflow.log)

linsenpago, 2022-10-10:
Hi! Thank you.. the new version works for me now! GPU is used now!

AdrianEddy, 2022-10-10:
@linsenpago thanks for testing!
@mickey1cx hmm, the d3dcompiler_47.dll is now included in gyroflow, and it's not working? and after you replace it with the other one from VS code, it's working? weird. Can you please also test other "Device for processing" under Advanced?

mickey1cx, 2022-10-10:
@AdrianEddy 
[wgpu] NVIDIA GeForce GTX 1050 Ti (Dx12) worked with replace dll (https://youtu.be/5Mi_WZCO6Eg)
:thumbsup: [wgpu] NVIDIA GeForce GTX 1050 Ti (Vulkan) worked with current version, without replacing

AdrianEddy, 2022-10-11:
@mickey1cx how about now in the dev build? https://gyroflow.xyz/devbuild/?autodownload I updated that dll to a newer one

closed 2022-10-11 by AdrianEddy