gyroflow/gyroflow #478: ubuntu 20.04.01 appimage not starting after update

minzukas, 2022-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.3.0-beta.2

### What operating system are you using?

ubuntu 20.04.01 LTS

### What GPU are you using?

Mesa Intel® UHD Graphics 620 (KBL GT2)

### What happened?

After Ubuntu upgrading Gyroflow appimage not starting.

### Relevant log output

```shell
mindaugas@tux:~/Downloads/gyroflow$ ./Gyroflow-linux64.AppImage
MDK 0.17.0 (git cc3146c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang15.0.2 GCC4.2.1; libc++15002; 12:29:34 Oct  9 2022
mdk plugin 'mdk-braw': 0x556ed36193b0
Crashdump written to /home/mindaugas/Downloads/gyroflow/0a844a5d-22f1-4984-55c68288-a26f6cbb.dmp
Segmentation fault (core dumped)
```

AdrianEddy, 2022-10-11:
did you try latest dev build? https://gyroflow.xyz/devbuild/?autodownload

minzukas, 2022-10-11:
Yes. just ran a new download and test. Still get the same

mindaugas@tux:~/Downloads/gyroflow(1)$ ./Gyroflow-linux64.AppImage
MDK 0.17.0 (git cc3146c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang15.0.2 GCC4.2.1; libc++15002; 12:29:34 Oct  9 2022
mdk plugin 'mdk-braw': 0x560d702263e0
Crashdump written to /home/mindaugas/Downloads/gyroflow(1)/406e562a-96a8-473d-00e75e9a-4fe2d23a.dmp
Segmentation fault (core dumped)

Does not allow me to attach.dmp

AdrianEddy, 2022-10-11:
did you try .tar.gz package?

AdrianEddy, 2022-10-11:
I just tried and AppImage works fine for me in Ubuntu. Did you install all necessary packages from the README? https://github.com/gyroflow/gyroflow/#linux

minzukas, 2022-10-11:
witth tar.gz I get 
error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory

will double check the packages

minzukas, 2022-10-11:
reinstalling the packages get something wrong with the drivers:

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package vdpau-va-driver is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  mesa-va-drivers:i386 mesa-va-drivers

AdrianEddy, 2022-10-11:
Install `mesa-va-drivers` instead of `vdpau-va-driver`

minzukas, 2022-10-11:
did not help, same issue

minzukas, 2022-10-11:
had some issues with the latest CURA appimage as well (openCL error), but reinstalled snap and downloaded linux-modern appimage which solved the problem.

AdrianEddy, 2022-10-11:
and Gyroflow v1.2.0 works for you?

minzukas, 2022-10-11:
Nope doesn't work any longer after prompted Ubuntu update. Used to work without problems previously though.

minzukas, 2022-10-12:
tar.gz launches after installing libc++-dev

Stampede10343, 2022-10-18:
I am also seeing a crash on Ubuntu 20.04.5

``01:00:39 [ERROR] thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value': /root/.cargo/registry/src/github.com-1ecc6299db9ec823/khronos-egl-4.1.0/src/lib.rs:961
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: <unknown>
  16: start_thread
             at /build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c:477:8
  17: clone
             at /build/glibc-SzIz7B/glibc-2.31/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Rayon: detected unexpected panic; aborting

AdrianEddy, 2022-10-18:
@Stampede10343 could you please check if latest dev build works for you? https://gyroflow.xyz/devbuild/?autodownload

Stampede10343, 2022-10-19:
> @Stampede10343 looks like your issue is the same as here #479 . Run with `QSG_RHI_BACKEND=vulkan`

Setting that does appear to fix the problem, thanks!

AdrianEddy, 2022-11-17:
This should be fixed by ceed75682c23f74245e80180bd409dd610f685e7, should work without any flags and no longer show an error in the log. Please check version gh871 https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-17 by AdrianEddy