gyroflow/gyroflow #479: Fedora 36 with RTX3060 - any version newer than 1.2.0 instant crash

ales-zima, 2022-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0-rc

### What operating system are you using?

Fedora Linux 36 (Workstation Edition) 64-bit

### What GPU are you using?

NVIDIA GeForce RTX 3060

### What happened?

Any newer version above 1.2.0 crash on start.
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9767414/gyroflow.log)

### Relevant log output

```shell
MDK 0.17.0 (git cc3146c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang15.0.2 GCC4.2.1; libc++15002; 12:29:34 Oct  9 2022
mdk plugin 'mdk-braw': 0x5573977be760
17:32:14 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.7.102") ["NVIDIA GeForce RTX 3060"]
17:32:14 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Portable Computing Language") Ok("OpenCL 2.0 pocl 1.8  Linux, RelWithDebInfo, RELOC, LLVM 13.0.1, SLEEF, DISTRO, POCL_DEBUG") []
17:32:14 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 22.1.7") []
17:32:14 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3060
17:32:14 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
17:32:15 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
17:32:15 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3060
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
17:32:15 [ERROR] thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value': /root/.cargo/registry/src/github.com-1ecc6299db9ec823/khronos-egl-4.1.0/src/lib.rs:961
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: <unknown>
  16: start_thread
  17: __clone3

Rayon: detected unexpected panic; aborting
fish: Job 1, './gyroflow' terminated by signal SIGABRT (Abort)
```

AdrianEddy, 2022-10-13:
Hmm, I can't reproduce this, I tested on Fedora 36 in a VM and it works fine for me. I had to install `dnf install libcxx-devel ocl-icd *vdpau*`
This crash seems to originate in egl when calling `eglMakeCurrent`, not sure what's the reason
Can you try running it under Vulkan? `QSG_RHI_BACKEND=vulkan ./gyroflow`

ales-zima, 2022-10-13:
Tried with QSG_RHI_BACKEND=vulkan - Gyroflow starts and running, but strange color flickering in video preview (color moire, artifacts).
Tried with NO_WGPU=1 - Gyroflow starts and running, everything seems to be OK. RTX3060 is seen as OpenCL and CUDA device.
NVIDIA proprietary driver used is 515.76. Your suggested packages installed (except libva-vdpau-driver, nvidia-vaapi-driver used instead).

ales-zima, 2022-10-17:
Tried on two different computers with Linux (Fedora 36). Intel based laptop with nVidia GTX1050 - same problem, running only with NO_WGPU. AMD based desktop with Radeon RX550 - runs OK. It seems like a problem between wgpu library and nVidia drivers.

ales-zima, 2022-10-18:
Tested with new version of Gyroflow (1.3.0 gh816) - exactly same behavior. Gyroflow crash instantly after start. With QSG_RHI_BACKEND=vulkan starts and is running, but image artifacts in preview window. With NO_WGPU=1 runs with no issues.

AdrianEddy, 2022-10-18:
Could you please attach gyroflow.log from both cases? with and without wgpu

ales-zima, 2022-10-18:
Here we go.

[gyroflow-NO_WGPU.log](https://github.com/gyroflow/gyroflow/files/9811884/gyroflow-NO_WGPU.log)
[gyroflow-plain.log](https://github.com/gyroflow/gyroflow/files/9811885/gyroflow-plain.log)
[gyroflow-QSG_RHI_BACKEND.log](https://github.com/gyroflow/gyroflow/files/9811886/gyroflow-QSG_RHI_BACKEND.log)

ales-zima, 2022-10-18:
With pipeline Zero-copy OpenCL instant crash. With pipeline OpenCL/wgpu/CPU same as pipeline Zero-copy Qt RHI - artifacts, color flickering.

[gyroflow_OpenCL-wgpu-CPU.log](https://github.com/gyroflow/gyroflow/files/9812623/gyroflow_OpenCL-wgpu-CPU.log)
[gyroflow_ZeroCopy-OpenCL.log](https://github.com/gyroflow/gyroflow/files/9812630/gyroflow_ZeroCopy-OpenCL.log)

ales-zima, 2022-10-18:
Tested with QSG_RHI_BACKEND=vulkan ofcourse.

ales-zima, 2022-10-18:
As I write before, with NO_WGPU=1 (only) Gyroflow run flawlessly. Best result for preview is given by Zero-copy Qt RHI. Slightly worse is OpenCL/wgpu/CPU pipeline and Zero-copy OpenCL is barely usable.

AdrianEddy, 2022-10-18:
Could you please test latest one (gh820) without flags?

ales-zima, 2022-10-18:
I can try, but where I get this version? Latest build is gh819 and still crashing.

AdrianEddy, 2022-10-18:
https://gyroflow.xyz/devbuild/?autodownload

ales-zima, 2022-10-18:
It's running. With no env variables. But it looks, that wgpu is still broken thus you are using fallback like NO_WGPU?
BTW: libmdk.so.0 is not included in archive.

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/9814124/gyroflow.log)

AdrianEddy, 2022-11-17:
This should be fixed by ceed75682c23f74245e80180bd409dd610f685e7, should work without any flags and no longer show an error in the log. Please check version gh871 https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-17 by AdrianEddy