gyroflow/gyroflow #480: Runcam Thumb Pro multiple file gyro sync issue

btandrei, 2022-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3 rc1

### What operating system are you using?

windows 10

### What GPU are you using?

M1000M

### What happened?

Hi, there's a bug with the Runcam Thumb Pro, If the video file is too long(>9min) it gets split between multiple files, then the gyro data gets out of sync on the next files, I don't know how to re-sync it, the first file stabilizes fine, this might also be a bug that comes from the camera.

ElvinC, 2022-10-14:
Could you attach some sample files with this issue?

mykls, 2022-10-17:
Similar issue with Runcam 5 Orange. Gyroflow can join the files, but then can't read the gyro data, so I have to stabilise each file individually, but then the transitions don't sync up when files are joined.

ElvinC, 2022-10-17:
Last build supports joining gcsv files, try with that: https://gyroflow.xyz/devbuild/
Note that from a quick test, there seems to be a slight offset in the order of a couple hundred milliseconds at the joining point (missing samples from the camera side), so you might need a couple of extra syncs around that point

mykls, 2022-10-18:
Thanks.  Tried the dev build and it linked the files, applied the lens settings, loaded the gyro data and commenced auto-sync, however each time (tried about 5 times) the application just shut-down at around 50-60% through the auto-sync.  File was 2.7k and was around 40 minutes long.

ElvinC, 2022-10-18:
Could you try runnning the program using `gyroflow_with_console.bat` and post the log output when it crashes?

mykls, 2022-10-18:
It was an 'out of memory' error.  Closing everything else managed to get me through the auto sync while using up to 16GB of RAM, however I then had the export fail to complete.  Maybe 40 minutes and 18GB of combined file is asking a little too much for my computer.

AdrianEddy, 2022-11-08:
You can try lowering "Processing resolution" in Synchronization -> Advanced

closed 2022-11-08 by AdrianEddy