gyroflow/gyroflow #481: Gyroflow v1.2.0 not opening on Windows 11 / RTX3070 and win10 rtx 2080ti

leesin597, 2022-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3

### What operating system are you using?

win11 22h2 and win 10 21h1

### What GPU are you using?

rtx 3070 laptop  and rtx 2080ti  (pc)

### What happened?

1.3 It can't run on my two computers, and all drivers and systems have been updated. Click to start the program and then crash

### Relevant log output

```shell
06:40:39 [INFO] [MDK] MDK 0.17.0 (git cc3146c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1933; MSVCRT14.33.31629.0/msvcp140.dll; 12:35:05 Oct  9 2022
06:40:39 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
06:40:39 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
06:40:39 [DEBUG] (1) mdk: [MDK] user module: C:\Users\lixue\Desktop\gyroflow\Gyroflow.exe
06:40:39 [DEBUG] (1) mdk: [MDK] user module: C:\Users\lixue\Desktop\gyroflow\Gyroflow.exe
06:40:39 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
06:40:40 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.7.99") ["NVIDIA GeForce RTX 3070 Laptop GPU"]
06:40:40 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") Ok("OpenCL 3.0") ["Intel(R) UHD Graphics"]
06:40:40 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3070 Laptop GPU
06:40:40 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
06:40:40 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
06:40:40 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3070 laptop gpu
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:40:40 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""

06:47:58 [INFO] [MDK] MDK 0.17.0 (git cc3146c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1933; MSVCRT14.33.31629.0/msvcp140.dll; 12:35:05 Oct  9 2022
06:47:58 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
06:47:58 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
06:47:58 [DEBUG] (1) mdk: [MDK] user module: D:\program\gyroflow\Gyroflow.exe
06:47:58 [DEBUG] (1) mdk: [MDK] user module: D:\program\gyroflow\Gyroflow.exe
06:47:58 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
06:47:59 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.7.102") ["NVIDIA GeForce RTX 2080 Ti"]
06:47:59 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2080 Ti
06:47:59 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
06:47:59 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
06:47:59 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2080 ti
06:47:59 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:47:59 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:48:00 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:48:00 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:48:00 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:48:00 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:48:00 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
06:48:00 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
```

AdrianEddy, 2022-10-13:
Does v1.2.0 work for you? 
Can you try adding `set NO_OPENCL=1` as a second linbe in `Gyroflow_with_console.bat` and then run this .bat file

leesin597, 2022-10-13:
1.2 Neither. When I add this, still can't
12:20:07 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "NVIDIA GeForce RTX 2080 Ti", vendor: 4318, device: 7687, device_type: DiscreteGpu, driver: "NVIDIA", driver_info: "517.40", backend: Vulkan }
12:20:07 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
12:20:07 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
12:20:07 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia geforce rtx 2080 ti
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
12:20:07 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""

thread 'main' has overflowed its stack
Crashdump written to D:\program\gyroflow/28892199-8a13-42e9-a66a-9f6686e88e59.dmp
请按任意键继续. . .

AdrianEddy, 2022-10-17:
How about v1.3.0? Or adding `NO_WGPU=1` to the bat file

AdrianEddy, 2022-11-18:
Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-11-21:
I believe this is either fixed in latest dev build or a driver issue, make sure you have all drivers updated and all windows updates installed, and if it still happens in latest dev build, please reopen https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-21 by AdrianEddy