gyroflow/gyroflow #482: Could you add implementation for the Vuze XR camera?

FloroVR, 2022-10-13:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Could you add implementation for the Vuze XR camera?

AdrianEddy, 2022-10-13:
Gyroflow doesn't support 360 deg cameras and it's not planned. If there's a mode that shoots regular video, then it could work, do you have a sample file?

FloroVR, 2022-10-13:
when i open the mp4 file in quicktime, it shows two video tracks, two audio
tracks and the attached track, which appears to include calibration
information and also encoded text that I think might be the IMU data.  I
extracted that track through quicktime and attached it to this email.
Here's a link for the full video file:
https://drive.google.com/file/d/1QrYfuTGKBgXotVrsCq5GcTsPi1Jdgfnh/view?usp=sharing
The front facing view in the video is video track 2.  I'm trying to at
least find a way to read the IMU data.  Since you said GyroFlow doesn't
stablize 360, or presumably any fisheye/180/equirectangular projection
videos, i'll figure out how to use the data to stabilize once i can read
it.  Any help would be greatly appreciated.

AdrianEddy, 2022-11-10:
Can you record another sample with this camera? I have the data format figured out but it's hard to figure out correct orientation with this file you provided. Can you record short handheld sample where you move the camera up/down, right/left and rotate?

AdrianEddy, 2022-11-10:
Initial implementation here https://github.com/AdrianEddy/telemetry-parser/blob/master/src/vuze/mod.rs
Latest dev build of gyroflow will load that data and embedded calibration data as lens profile automatically.

closed 2022-11-18 by AdrianEddy