gyroflow/gyroflow #483: 1.3.0-rc - no drag and drop possible if started as administrator under Win11

linsenpago, 2022-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0-rc

### What operating system are you using?

Win11

### What GPU are you using?

Sapphire Radeon RX470

### What happened?

If i started Gyroflow as administrator drag & drop video files is not possible.

Bug or other reason?

AdrianEddy, 2022-10-14:
This is how windows works, you can't use drag and drop between normal and elevated programs. We can't do anything about it, it's a security measure by Windows

closed 2022-10-14 by AdrianEddy

linsenpago, 2022-10-14:
Oh! Sorry! Did not know that! 🙈