gyroflow/gyroflow #485: Lens profile disappeared after failure to export

lillianhorin, 2022-10-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.3.0-beta

### What operating system are you using?

Windows 10

### What GPU are you using?

Intel(R) HD Graphics 530, NVIDIA Quadro M1000M

### What happened?

Thank you for developing this wonderful software. I just started using it and things were going smoothly - at first. I found the correct lens profile (Sony ZV-E10, 4k, Sigma 30mm f1.4, 29.97fps) and it corrected the video beautifully. Then upon trying to export, I got an error (40:function not implemented, NVENC not compatible) which caused the software to crash. Upon crashing, the lens profile that was correct for my camera disappeared. I tried re-installing, restarting my computer, and using similar picture profiles and nothing can replicate the results of the original.

Any help in these two issues would be appreciated:

- Recovering the original lens profile
- Solving the export issue ** especially this one! 

Unfortunately I can't replicate the original issues because the lens profile has disappeared, but I am happy to provide you with anything else that might be helpful.

**Export Error**

[image]

**Log**
Just to add on, for the sake of troubleshooting the export issue, I picked an arbitrary lens profile and here is the error.

### Relevant log output

```shell
01:18:51 [INFO] [MDK] MDK 0.17.0 (git 347aa9b) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1933; MSVCRT14.33.31629.0/msvcp140.dll; 15:32:58 Sep 30 2022
01:18:51 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
01:18:51 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
01:18:51 [DEBUG] (1) mdk: [MDK] user module: C:\Users\lilli\Desktop\Applications\Gyroflow\Gyroflow.exe
01:18:51 [DEBUG] (1) mdk: [MDK] user module: C:\Users\lilli\Desktop\Applications\Gyroflow\Gyroflow.exe
01:18:51 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
01:18:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.6.134") ["Quadro M1000M"]
01:18:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL") Ok("OpenCL 2.1") ["Intel(R) HD Graphics 530"]
01:18:52 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_khr_gl_event cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_nv_kernel_attribute cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation Quadro M1000M
01:18:52 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
01:18:52 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
01:18:52 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation quadro m1000m
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\lilli\\Desktop\\Applications\\Gyroflow\\camera_presets"
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
01:18:52 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
[... 448 lines cut ...]
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
01:20:31 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PreConversion
01:20:31 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 8008, render_frame_count: 240
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_CUDA)
01:20:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x19efff2af00
01:20:31 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
01:20:32 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
01:20:32 [DEBUG] (2) gyroflow::rendering: Possible encoders with Nvidia: [("hevc_nvenc", true), ("hevc_mf", true), ("libx265", false)]
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
01:20:32 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PreConversion
01:20:32 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 8008, render_frame_count: 240
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_CUDA)
01:20:32 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x19ee858c2c0
01:21:12 [WARN] Retrying to obtain clipboard.
```
```

AdrianEddy, 2022-10-16:
This error means that your GPU doesn't support hardware encoding, you can try to use the Intel device (under Advanced in export settings) or uncheck GPU encoding.

As for the lens profile, is it this one? https://github.com/gyroflow/gyroflow/blob/master/resources/camera_presets/Sony/Sony_ZV-E10_SIGMA%2030MM%20F1.4%20ND__4k_16by9_3840x2160-23.98fps.json  You can download it from there and load manually

lillianhorin, 2022-10-16:
Thank you so much for the quite response! It's so appreciated.

> This error means that your GPU doesn't support hardware encoding, you can try to use the Intel device (under Advanced in export settings) or uncheck GPU encoding.

So I tried every possible combination of checking/unchecking GPU encoding and every graphics device (see image below) including "CPU only" and keep getting the same error. I've used my GPU to encode videos before so not quite sure what's going on here.

[image]

> As for the lens profile, is it this one? https://github.com/gyroflow/gyroflow/blob/master/resources/camera_presets/Sony/Sony_ZV-E10_SIGMA%2030MM%20F1.4%20ND__4k_16by9_3840x2160-23.98fps.json You can download it from there and load manually

Close! The fps rate is different.. it should be ~29 fps.

lillianhorin, 2022-10-16:
Just wanted to update this! I realized I unchecked the wrong box... I unchecked "Use GPU decoding" which is also under Advanced, but not under export settings. I found the proper box ("Use GPU encoding") and unchecking that solved the problem. Thank you so much for your help!

For anyone who has this issue in the future and is looking through this, in my specific situation it looks like the H.265 export was specifically the issue. I can use GPU encoding with H.264 no problem. For H.265 export, I do have to uncheck GPU encoding.

One thing I wanted to note as an additional bug. Whenever I open GyroFlow, it defaults to the Intel driver. With these new settings if I open any video the program will automatically crash. Now that I know to switch drivers, the problem is fixed, but it is pretty consistent that it causes crashes by defaulting to that driver.

AdrianEddy, 2022-10-16:
Please test latest dev build https://gyroflow.xyz/devbuild/?autodownload .

Also note that there are two different settings for device: Device for rendering (the video encoder) and device for processing (for the stabilization process). You may have some luck with Intel encoding and Nvidia processing.

As for the lens profile: fps doesn't matter in this case and we didn't have a different one so it had to be the one I sent you.

closed 2022-10-16 by AdrianEddy