gyroflow/gyroflow #487: PC crashes on exporting using GPU

avib99, 2022-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.3.0

### What operating system are you using?

Win 10

### What GPU are you using?

RTX 2080Ti

### What happened?

PC crashes while exporting using GPU, causing a black screen and pc fans running full blast.

AdrianEddy, 2022-10-18:
Do you have latest GPU drivers installed and latest windows updates? Does it happen with all export codecs?

AdrianEddy, 2022-11-11:
Please include gyroflow.log file (created next to Gyroflow.exe) after it crashes

AdrianEddy, 2022-11-21:
I believe this is fixed in latest dev build, please reopen if it still happens https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-21 by AdrianEddy