gyroflow/gyroflow #490: gyroflow crashes when attempting to export - MAC OS

cjvisions, 2022-10-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0

### What operating system are you using?

monterey 12.6

### What GPU are you using?

Intel Iris Pro 1536 MB

### What happened?

every time i attempt to export my video, gyroflow crashes.  It was working perfectly fine until i downloaded the latest update 1.3.0...now none of the versions work and all crash.

AdrianEddy, 2022-10-19:
Did you try setting different device for processing under Advanced?

AdrianEddy, 2022-10-19:
Please also include gyroflow.log file (Show package contents -> Contents -> MacOS -> gyroflow.log) after it crashes

cjvisions, 2022-10-20:
It wont allow me to choose any other device other than apple iris pro: 1.2  for rendering in advanced.  Ive been trying to change around different settings snd nothing seems to stop it from crashing.Carlos Galindo

AdrianEddy, 2022-10-20:
Please include gyroflow.log file (Show package contents -> Contents -> MacOS -> gyroflow.log) after it crashes