gyroflow/gyroflow #491: stabilization based on selection

VBK7, 2022-10-20:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

When I select a certain portion of a clip with a big shake right after the end of the selected section of video, even though the shake is not selected, Gyroflow starts zooming in towards the end of the selection even though the shaky portion is not selected and I don't need it but everything right up to the big shake is very important and I need as wide of a frame as possible. Would be nice to have an option to turn on and off whether Gyroflow takes into consideration gyro data outside the selected portions of clips. I believe I didn't have this issue on older versions.

AdrianEddy, 2022-11-09:
Done in f216320e3dbeacd1222ee933302efc66d06c96ad, use latest dev build https://gyroflow.xyz/devbuild/?autodownload

In `Stabilization -> Advanced` check "Only within trim range" 

[image]

closed 2022-11-09 by AdrianEddy

VBK7, 2022-11-10:
@AdrianEddy Great! This doesn't seem to work with the simple gyro integration method though. Is this feature specific to certain integration methods currently?

AdrianEddy, 2022-11-10:
Integration method doesn't matter, but anyway you shouldn't really use "Simple gyro", use either "Complementary" or "VQF"

I tested it and it worked as expected for me, make sure your trim range is accurate and doesn't contain any sample of the rapid movement. If that still doesn't work for you then you can send me your video and project file

sidorykp, 2025-01-29:
I have the same issue as the author of this issue but I use BRAW files. So I cannot use Gyroflow to export the stabilisation result. I have to use the OFX plug-in for this purpose but it does not have the "Only within trim range" feature.
Any chance that the OFX plug-in will have the "Only within trim range" feature?

sidorykp, 2025-01-29:
Yes, it works indeed. I used an outdated OFX plug-in version. Thank you.