gyroflow/gyroflow #492: Discontinuity of stabilization result.

S-o-T, 2022-10-23:
Stabilization of gopro8 recording with "Default" (with "Static zoom") preset leads to the video accessible [here](https://drive.google.com/drive/folders/1XanN5nvC24qiHyEzVCvTkZV3Iua80LO2?usp=sharing). There is clearly (look at the tree, also video with extended fov provided) a problem with "jerky" movement (jumps/discontinuity of virtual camera frustum). I am interested whether it is due to the nature of camera motion/imu data recorded, is it due to some unfavorable settings selected, is it due to specifics of "stabilization algorithm" or is it simply a bug somewhere?

AdrianEddy, 2022-10-24:
This video has skipped frames at these times, I've seen this one before and it's a camera firmware issue. Unfortunately there's nothing we can do about it.

closed 2022-10-24 by AdrianEddy

S-o-T, 2022-10-24:
Thanks for the answer. Can you please share the method you used to decide if original video has skipped frames? It would be useful to automate such detection and give a verbose warning to the user, so the user wont spend his time trying to tune stabilization settings.