gyroflow/gyroflow #493: Very shaking result on gopro6 EIS enabled videos

yzkzero, 2022-10-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.2 & 1.3

### What operating system are you using?

Win10

### What GPU are you using?

GeForce GTX 1050

### What happened?

Senario: Gopro Hero6 Black mounted on bicycle
Camera parameters: 2K 16:9 wide 30fps, EIS enabled (or 4K 16:9 superview with EIS enabled)
Gyroflow parameters: all default

Bug: Gyroflow generates very shaking video compare the to original one (even worse than those gopro EIS disabled video)

If I change the camera parameters to 4K 4:3 wide 30fps, **EIS disabled**. Then the result is much better which is like gopro EIS enabled video. But I don't want to disable EIS on gopro for all videos.

Any idea? Is gyroflow calibration needed to generate a correct video in my case? Or EIS must be off on gopro6?

Thanks a lot!

yzkzero, 2022-10-28:
In ReelsteadyGo Q&A document, I found it says: "You'll need to turn all stabilization features off in your camera's options/preferences." 
But in gyroflow document, I don't find such description. It just recommends using 4:3 resolution.

AdrianEddy, 2022-10-28:
Hero 6 doesn't work with EIS enabled because it doesn't record necessary data for that stabilization. Hero 8 is the first GoPro to support it

closed 2022-10-28 by AdrianEddy