gyroflow/gyroflow #495: Errors after pressing the rendering button

ObiVanDunobi, 2022-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.3.0

### What operating system are you using?

Win10

### What GPU are you using?

NVidia GeForce GT 720M

### What happened?

Hi,
After pressing the export button, the rendering process does not start, but several errors pop up (see in the attachment).
Does ffmpeg need to be started manually somewhere?
Does NVidia error mean my graphics card is not supported? I think I have the latest driver.
Is there any other way to run Gyroflow without using a graphics card? For example CPU?
I got Intel Core i5-3337U 1.80GHz.
Or am I out of luck? :)
So far I've been using Gyroflow 0.3.0 beta and even though it took hours and threw some red messages during rendering, I was happy with the result.
Is there any chance I can use v1.3.0, or should I be satisfied with at least v0.3.0 with my hardware?

[image]

Thank you

AdrianEddy, 2022-11-04:
NVidia GeForce GT 720M doesn't support hardware rendering, simply uncheck GPU encoding

closed 2022-11-04 by AdrianEddy

ObiVanDunobi, 2022-11-06:
Oh, that simple.
Now works.
THX

ObiVanDunobi, 2022-11-06:
> NVidia GeForce GT 720M doesn't support hardware rendering, simply uncheck GPU encoding

Oh, that simple.
Now works.
THX