gyroflow/gyroflow #497: Latest build (1.3) not running on Ubuntu 22.04 LTS

softwarelifecycle, 2022-11-07:
Hello all:
My preferred OS is Ubuntu and the only thing that I need in my sweet of tools that doesn't work on it is Gyroflow.
It works fine on Windows....but I do not like window!

I downloaded the Gyroflow-linux64.Appimage and upon running all it does is produce a .dmp file...
For the prior version of Gyroflow, I followed these instructions:
https://github.com/gyroflow/gyroflow/issues/154#:~:text=I%20can%20confirm%20that%20following%20the%20steps%20linked
And they worked... but they do not work for the latest 1.3 build.

This should not be so difficult... Are there any other approaches that work? Can the dmp file be uploaded here?

Thanks
Harry

AdrianEddy, 2022-11-07:
Use latest dev build and use .tar.gz package instead of .AppImage: https://gyroflow.xyz/devbuild/?autodownload

angus-elsa, 2022-11-07:
1.3 not working on 20.04 either.

AdrianEddy, 2022-11-07:
> 1.3 not working on 20.04 either.

did you check the latest dev build from the link I provided?

AdrianEddy, 2022-11-08:
Duplicate of #478

closed 2022-11-08 by AdrianEddy

angus-elsa, 2022-11-08:
> > 1.3 not working on 20.04 either.
> 
> did you check the latest dev build from the link I provided?

Worked perfectly my friend, cheers buddy