gyroflow/gyroflow #498: Gyroflow CLI commands are not syncronous

kernel-machine, 2022-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0

### What operating system are you using?

Windows 11

### What happened?

Gyroflow has some problems with the CLI implementation, i have found this problem when i have starting to write a bash script (for Windows PowerShell) to stabilize my videos and then when the stabilization is ended i would like to use the ffmpeg commands to rotate the videos.

The problem is that when i execute a gyroflow command in the CLI, it immediately exits and the output will be printed after the exit phase.

For example, if i write a .ps1 script like this:
```
gyroflow.exe my_video.mp4 -g my_data.gcsv
ffmpeg.exe -i my_video_stabilized.mp4 (set some parameters)
```
Gyroflow starts the stabilization, since the command is not syncronous, it "exits" immediately and the powershell execute the next command (the one with ffmpeg), now ffmpeg cannot take in input the stabilized video, because gyroflow is still processing it, so ffmpeg exits, while gyroflow continues to print the progresses.

I hope that my explanation is clear

kernel-machine, 2022-11-07:
For example another issue linked to the previous one is that when for example i enter "gyroflow.exe --help"
I get this output and seems like gyroflow never teriminates
[image]
Because i don't have `PS C:\Users\Luca>` at the end of gyroflow output

AdrianEddy, 2022-11-07:
This is not a problem with Gyroflow, but with the way you start commands in Powershell. If you do the same commands using .bat file, it all works correctly.
If you want to execute command in powershell and wait for it to finish, you need to call it using `Start-Process -Wait`, see more here: https://stackoverflow.com/questions/1741490/how-to-tell-powershell-to-wait-for-each-command-to-end-before-starting-the-next

Your second issue is also just because of Powershell prompt. Gyroflow finishes, you can see that in process list. Simply click Enter to see your prompt again, or start typing your next command

closed 2022-11-07 by AdrianEddy