gyroflow/gyroflow #499: "Open File" file dialog crashes app in latest build

softwarelifecycle, 2022-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0 (gh836)

### What operating system are you using?

Ubuntu 22.04.1 LTS

### What GPU are you using?

NVIDIA GeForce RTX 2070 Rev. A

### What happened?

- Installed the latest nightly build 1.3.0 (gh836)
- Clicked on Open File button
- Immediately crashed
[image]
It does open up the file if dragged onto the UI

AdrianEddy, 2022-11-15:
Looks like an issue in GTK: https://www.google.com/search?q=filechooser+does+not+contain+%22show-type-column%22
did it work in previous gyroflow version?

AdrianEddy, 2022-11-17:
Is this AppImage or .tar.gz?

AdrianEddy, 2022-11-18:
I believe this is fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-18 by AdrianEddy

softwarelifecycle, 2022-11-18:
It was in the Appimage.. that's been the one that worked the best... I never saw it work... trying out now the new version...

softwarelifecycle, 2022-11-18:
ok, the latest build fixes the open dialog bug! :)   Thanks very much!
Still getting the green screen... Adding comment there...