gyroflow/gyroflow #500: Video Preview turns green during playback

softwarelifecycle, 2022-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0 (gh836)

### What operating system are you using?

Ubuntu 22.04.1 LTS

### What GPU are you using?

NVIDIA GeForce RTX 2070 Rev. A

### What happened?

1. Dragged mp4 file from DJI Avata into UI
2. Loaded fine and showed gyro data
3. Played video and was ok for about 10 seconds then screen turned green!
[image]
4. I expected the video to not turn green and continue playing... On another video it did make it all the way through.

AdrianEddy, 2022-11-09:
Most likely driver issue, do you have latest updates installed?

softwarelifecycle, 2022-11-15:
Hi Adrian,
I just updated my drivers for the NVIDIA card to the latest 520.56 version and same problem... plays OK for a bit then turns green like above...

AdrianEddy, 2022-11-15:
you can try setting `export MDK_DECODERS=ffmpeg` in terminal before launching

softwarelifecycle, 2022-11-15:
ok i'll try that too, thanks.. it's 1am here now. :)
I also had just downloaded the latest daily build for today and it gave a solid green preview after about 10 seconds instead of that pattern above... and it does export a clean video.. so only in preview still
Thanks very much! It's doing a great job overall stabilizing my Avata footage.

softwarelifecycle, 2022-11-16:
I tried that export command first but it then wouldn't allow me to drag a file into the previewer panel like i was doing... I'd drag a file on and just get the spinner... But just before running the export, it was fine. then when I started up a new instance in a new command window, it was also fine... so the export command broke something... :(

AdrianEddy, 2022-11-16:
please upload gyroflow.log file

softwarelifecycle, 2022-11-18:
[AfterExportLog.txt](https://github.com/gyroflow/gyroflow/files/10036909/AfterExportLog.txt)
This is the output from Gyro in the terminal window.. no separate log file was created.
I had first entered the export command as you stated above then ran Gyro and tried to drag a file onto the previewer and it just hung for like 10 minutes till I terminated it...

AdrianEddy, 2022-11-18:
Please test the .tar.gz package from latest dev build and make sure you have installed all recommended packages from the README https://github.com/gyroflow/gyroflow/#linux

https://gyroflow.xyz/devbuild/?autodownload

softwarelifecycle, 2022-11-18:
[NoExport-gyroflow.log](https://github.com/gyroflow/gyroflow/files/10037623/NoExport-gyroflow.log)
ok, this is the log file without running that export command you mentioned above...It ran in preview pretty far and turned green with about 10 seconds to go.

When I did use the export command and dragged a file into the viewer it did that same hang as before... just the spinner...

Hope this helps!
Thanks for your time on this!
Harold

AdrianEddy, 2022-11-18:
@wang-bin, there are some MDK errors in this log, could you please check if there's anything we can do?

softwarelifecycle, 2022-11-24:
Anything able to be done about this?

wang-bin, 2022-11-24:
Can you try the command `ffmpeg -an -hwaccel vdpau -i your_video -f null >/dev/null`

softwarelifecycle, 2022-11-27:
[image]
Here is the output after running the ffmpeg command from above.  What does this show?

softwarelifecycle, 2022-11-27:
I see there is a line towards the top about a "Library Configuration Mismatch"
Just to clarify again.. this always seems to export ok.. but turns green in preview.

softwarelifecycle, 2022-12-02:
Is there anything useful in the log output in the last post?

wang-bin, 2022-12-02:
from gyroflow log, the first frame is decoded. vdpau is recreated after a seek and failed. system ffmpeg version is different with gyroflow's, so it's hard to tell. You can download https://sourceforge.net/projects/mdk-sdk/files/mdk-sdk-linux.tar.xz/download and run `./mdk-sdk/bin/amd64/glfwplay -c:v VDPAU yourvideo.mp4`, if video shows, click on the window and video will seek to a new position, and see what happens. If turns green, delete mdk-sdk/lib/amd64/libffmpeg.so.5 and run again, then system ffmpeg will be used, if still green, then there's is a bug in my code.

AdrianEddy, 2022-12-09:
@softwarelifecycle any update?

softwarelifecycle, 2022-12-25:
Sorry, iam trying now.. was on another project for a bit...

softwarelifecycle, 2022-12-26:
I tried the above and get the following error:  The lib file being referenced I could not locate anywhere

[image]

AdrianEddy, 2022-12-26:
try `sudo apt install libglfw3`

softwarelifecycle, 2022-12-26:
I installed that library and it worked... I viewed 3 different videos and they all stayed normal... no green screen like above... 

I then also renamed that libffmpeg.so.5 so it couldn't find it and it also played the video fine... 
So only seems to be happening within the Gyroflow environment... What do you think?

AdrianEddy, 2022-12-26:
did you try to seek the video to a new position when playing with glfw play?

softwarelifecycle, 2022-12-26:
oh yeah.. I clicked around a lot and that was fine....

softwarelifecycle, 2022-12-28:
Do you think this is an issue with Gyroflow?

dotSynergy, 2023-01-14:
getting the same problem:

- Have you tried the latest build?

    yes

- Gyroflow version

1.4.2 (gh972)

- What operating system are you using?

Pop!_OS 22.04 LTS

- What GPU are you using?

NVIDIA GeForce RTX 1070

- What happened?

After a couple seconds of preview the image goes green. If i fiddle with the window (resize, enlarge preview) the video preview comes back fine, but stops after a couple tries. Video is from a DJI Action 2.

All the suggested packages are installed, even libglfw3.
Starting with `export MDK_DECODERS=ffmpeg` makes the program hang when i try to import a video. ffmpeg test runs fine. glfwplay test runs fine.

[image]

AdrianEddy, 2023-01-14:
Please include gyroflow.log file that is created after closing the program (after it shown green frames). The log in console doesn't contain all necessary information

dotSynergy, 2023-01-14:
sure, here it is:

https://dotsynergy.lol/filebrowser/share/AWWcSuEu

gyroflow.log and crashdump upon app losing

AdrianEddy, 2023-01-14:
Hmm, I'm seeing some audio errors, could you try muting the playback (with the button below the video)? Not sure if it's related but it will be good to know

dotSynergy, 2023-01-14:
> Hmm, I'm seeing some audio errors, could you try muting the playback (with the button below the video)? Not sure if it's related but it will be good to know

no difference, audio muted or unmuted. i'm trying right now on a different audio output, but so far the issue remains.

dotSynergy, 2023-01-14:
the green screen is a little different though:

https://dotsynergy.lol/filebrowser/share/kUm_CMTq

AdrianEddy, 2023-01-15:
> sure, here it is:
> 
> https://dotsynergy.lol/filebrowser/share/AWWcSuEu
@wang-bin hmm do you see anything in this log? It definitely looks like it comes from MDK, because the green view is moving around correctly, so Gyroflow is processing the green output from the video decoder

it's weird that it works in glfwplay though, maybe it's a different decoder or something?

@dotSynergy please also test with 
1. `./mdk-sdk/bin/amd64/glfwplay -c:v CUDA -logfile cuda.log yourvideo.mp4`
2. `./mdk-sdk/bin/amd64/glfwplay -c:v VDPAU -logfile vdpau.log yourvideo.mp4`
3. `./mdk-sdk/bin/amd64/glfwplay -c:v VAAPI -logfile vaapi.log yourvideo.mp4`
4. `./mdk-sdk/bin/amd64/glfwplay -c:v FFmpeg -logfile ffmpeg.log yourvideo.mp4`
5. Gyroflow with `export MDK_DECODERS=FFmpeg`
6. Gyroflow with `export MDK_DECODERS=CUDA`
7. Gyroflow with `export MDK_DECODERS=VDPAU `
8. Gyroflow with `export MDK_DECODERS=VAAPI`

and please send all these log files, it will help us figure out what's happening

dotSynergy, 2023-01-15:
sure, here's results:

1. black screen
2. everything fine
3. black screen
4. jumpy/laggy playback plus audio out of sync but otherwise fine
5. same as 4
6. loads video indefinitely, audio starts but with heavy stuttering
7. goes green after 20 seconds
8. same as 6

logs here inside "tests" folder:
https://dotsynergy.lol/filebrowser/share/AWWcSuEu

wang-bin, 2023-01-15:
> sure, here's results:
> 
> 1. black screen
> 2. everything fine, whole video
> 3. black screen
> 4. jumpy/laggy playback plus audio out of sync but otherwise fine
> 5. same as 4
> 6. loads video indefinitely, audio starts but with heavy stuttering
> 7. goes green after 20 seconds
> 8. same as 6
> 
> logs here inside "tests" folder: https://dotsynergy.lol/filebrowser/share/AWWcSuEu

vaapi is using nvidia driver and initialize error. you can run `vainfo` paste the output here. There are other drivers you can try, for example `nouveau` for nvidia, `i965` for intel, all the drivers are in `/usr/lib/x86_64-linux-gnu/dri/`. You can try another driver by setting environment var `export LIBVA_DRIVER_NAME=nouveau`, and `${LIBVA_DRIVER_NAME}_drv_video.so` in the driver dir will be used.

cuda fails because something is not enabled in my linux build, i can fix it soon.

vdpau log seems fine. I can't tell why goes green. playback paused after 20 seconds in the log.

dotSynergy, 2023-01-15:
> you can run vainfo paste the output here.

here you go:
```
~> vainfo
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```

strange because the nvidia driver works fine, i've been exporting videos with nvenc on OpenShot plus gaming without issues. nvidia-settings:
[image]
```~> nvidia-smi -L
GPU 0: NVIDIA GeForce GTX 1070 (UUID: GPU-7033a8f6-f328-b120-81bd-1ef13122ab4a)```

AdrianEddy, 2023-01-15:
Could you also try in Gyroflow if playback with stabilization off (the toggle button below the video) also turns green? Or with stab on but different "Preview pipeline " under Advanced

wang-bin, 2023-01-15:
> strange because the nvidia driver works fine, i've been exporting videos with nvenc on OpenShot plus gaming without issues. nvidia-settings:

vaapi is a standard api on linux, maybe nvidia driver is not well implemented. nvenc is nvidia's own codec, it must work

dotSynergy, 2023-01-15:
> Could you also try in Gyroflow if playback with stabilization off (the toggle button below the video) also turns green? Or with stab on but different "Preview pipeline " under Advanced

just tried, goes green

/usr/lib/x86_64-linux-gnu/dri/ content:
```
~ [3]> ls /usr/lib/x86_64-linux-gnu/dri/
crocus_dri.so      nouveau_drv_video.so   swrast_dri.so
i965_drv_video.so  r300_dri.so            virtio_gpu_dri.so
iHD_drv_video.so   r600_dri.so            vmwgfx_dri.so
iris_dri.so        r600_drv_video.so      zink_dri.so
kms_swrast_dri.so  radeonsi_dri.so
nouveau_dri.so     radeonsi_drv_video.so
```

wang-bin, 2023-01-15:
@dotSynergy do you have amd gpu? radeonsi is amd driver

dotSynergy, 2023-01-15:
> @dotSynergy do you have amd gpu? radeonsi is amd driver

nope, it's a vm with only an nvidia passed through

AdrianEddy, 2023-01-15:
Maybe something related to Qt RHI specifically. I would have to prepare the qmlrhi play example to test

dotSynergy, 2023-01-15:
might be an issue in how popos handled their nvidia driver metapackage, i'll try to tinker with drivers. i recall i had to turn off something related to drm because there was no video.

wang-bin, 2023-01-15:
@dotSynergy can you test the linux build from https://github.com/wang-bin/mdk-sdk/actions/runs/3923930219

download and replace libmdk.so.0, then try CUDA again

dotSynergy, 2023-01-15:
> @dotSynergy can you test the linux build from https://github.com/wang-bin/mdk-sdk/actions/runs/3923930219
> 
> download and replace libmdk.so.0, then try CUDA again

it works, been running for a while and still no green screen

AdrianEddy, 2023-01-16:
@dotSynergy and @softwarelifecycle please test latest dev build https://gyroflow.xyz/devbuild/?autodownload
It should include the latest fixes in mdk now

wang-bin, 2023-01-16:
nvidia vaapi driver is not feature complete, it does not support vaapi rendering, which is required by 0-copy rendering in x11. drm seems broken too. so only vaapi copy mode works.

softwarelifecycle, 2023-01-16:
I just ran the appimage file and it seems to work great now, no green screen! :)

Thanks very much for the hard work on this!

What was the main issue?

AdrianEddy, 2023-01-18:
Gyroflow tries to use CUDA decoder by default on linux, and something was not enabled in mdk for that. I'm not sure what exactly it was, but I'm glad it works now :)

closed 2023-01-18 by AdrianEddy