gyroflow/gyroflow #501: Cannot properly save a .gyroflow file after edit

User-3090, 2022-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

lastet dev of today

### What operating system are you using?

Win10

### What GPU are you using?

3090

### What happened?

When dragging a gyroflow file into gyroflow there is no option to save changes to this file. Instead one can only re export with default name, which is different from the actual gyroflow opened. Behavior is different when dragging a gyroflow file into the render queue. Then there is a way to edit and save the actual Gyroflow file. Please make the two options equal.

AdrianEddy, 2022-11-09:
Note that "Save" button when editing render queue item only saves in the render queue, in memory. It doesn't modify the original .gyroflow file you loaded

User-3090, 2022-11-09:
I just figured this out after posting. And I have to ask, why? Can we get a workflow where it allows us to save an opened gyroflow file to it's original name? I often have many trim ranges per clip so many gyroflow files per clip. Editing them later is sort of painful, because export -> overwrite actually does not overwrite the opened file, but the one with the clip basename.

User-3090, 2022-11-09:
I'd love see a behavior where a save button always updates the actual opened gyroflow file, be it opened from render queue or not.

User-3090, 2022-11-09:
I have to go through 3000-4000 clips with gyroflow now and at this point workflow issues really become visible. I'll file separate reports about the OFX which I'd love to use, but due to Resolve processing order couldn't find a workflow at this scale.

AdrianEddy, 2022-11-09:
depending on what you want to do, you may be able to use CLI. If you want to modify certain setting in all .gyroflow files, you may be able do that using CLI's option `export_project`, `preset` and .gyroflow file as input

User-3090, 2022-11-09:
Thanks, I'll look into these. Still do you think having a Save button that actually saves the opened gyroflow file would be sensible to have?

AdrianEddy, 2022-11-09:
Sure, simply nobody asked for it before and I can't predict all possible features by myself

User-3090, 2022-11-10:
Thanks man. I didn't mean to sound pushy or anything. I'm just exited to start a larger non drone project with Gyroflow at its core.

I think as an end user I'd like to see a Save as soon as I open a gyroflow file, whether from render queue or not, which updates the file I opend with my changes. Physically updating the file and not just in memory also makes it easy and consistent to work with the OFX.

AdrianEddy, 2022-11-11:
It's done in 0cd1919790c9bffc3c669181c31acee95984959c 
- When loading project file now you have an option to `Save project file`
- When editing render queue item and it's a project file, on `Save` it will update the project file
- `Ctrl+S` will save the current project file, or create one (like `Export project file`)

closed 2022-11-11 by AdrianEddy

User-3090, 2022-11-13:
So cool, thanks!