gyroflow/gyroflow #502: Please add Yes/Rename/No ALL button in render queue

User-3090, 2022-11-09:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

It's a pain to click each option for many hundred files.

[image]

User-3090, 2022-11-10:
A hidden "don't ask for overwrite confirmation" flag available in default.gyroflow would help a lot meanwhile :) Thanks for your work and time.

AdrianEddy, 2022-11-10:
Done in e3dfd4782c03bcfafdab5fd8bb6bf0729dfc58f4 https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-10 by AdrianEddy

User-3090, 2022-11-13:
You're such an amazing dev. I just though about how to best implement it UI wise and that we need a way to specify the default export mode as well and you just nailed it overnight.

Thanks man!

[image]