gyroflow/gyroflow #504: Gitup F1 implementation possible?

MR444, 2022-11-12:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

The Gitup F1 is old but still still my favourite flat-shaped 4k DVR camera for planes. Is an implementation possible? How to get the gyro data from the camera? 
There are already modded firmwares available but could not find any with gyro data:
https://www.rcgroups.com/forums/showthread.php?2929991-GitUp-F1-%2A%2A%2Aowners-thread%2A%2A%2A&p=42025163&highlight=download#post42025163

AdrianEddy, 2022-11-13:
Do you have any information that this camera records gyro data? if it's not stated anywhere then it probably doesn't, it's not a standard feature. If you are confident that this camera records gyro data and you have a sample file, then I can implement support in Gyroflow.

MR444, 2022-11-16:
I contaced the customer support and some of the modded firmware devs but no-one seem to be interested in this camera anymore. So I think the gyro data are not recorded. How can I check if there are any gyro data in the video file?

AdrianEddy, 2022-11-16:
So I checked the files and there doesn't seem to be anything that resembles gyro data in these files. There is some kind of additional binary data in the `free` section of the mp4 file, but it looks totally random and it's unlikely there's any gyro data in there. It's impossible to decode it without the documentation or sources of the writing library, so there's nothing I can do here.

closed 2022-11-16 by AdrianEddy

MR444, 2022-11-16:
Thank you for your checks. I don't think the manufacturer will take care and develop a new firmware with gyro data so I will switch to another camera.