gyroflow/gyroflow #509: Synching multiple seggments Runcam Thump Pro

helicocrasher, 2022-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.3.0

### What operating system are you using?

Betriebsystemname	Microsoft Windows 11 Pro Version	10.0.22000 Build 22000

### What GPU are you using?

Prozessor	Intel(R) Core(TM) i7-10510U GPU 

### What happened?

I record very long rc plane flights with Runcam Thumb Pro in loop recording (set = 5 min).
If I stabilize single segments of such recordings the all have a stable offset in the +1500ms range.
If I join and stabilize the multiple segments the offset is incrementing with every new segment by about 300ms. 
This requires to add autosynch points shortly before and shortly after the 5min borders. 
This works OKish but is time consuming work and there remains some shakes between these synch points.  

https://github.com/gyroflow/gyroflow/issues/480 adressed a similar issue but was closed rather unconclusive.

AdrianEddy, 2022-12-02:
Can you also upload the original source video files? Or the joined file if you deleted the original ones

AdrianEddy, 2022-12-06:
Added these additional sync points in the latest dev build
You need to re-join the files, so Gyroflow knows where are the joining points
https://gyroflow.xyz/devbuild/?autodownload

closed 2022-12-06 by AdrianEddy

helicocrasher, 2022-12-06:
@AdrianEddy 
Sorry for not answering eralier.. 
and thank you the for the dev build.
Just been testing. 
I see the additional syncpoint inserted near the 5min splice ... cool!
unluckily the the second syncpoint is too much forward on the time line still in 5m00s range and so it does not help it results at the same offset as the previous syncpoint. If I manually place an autosync, I need to go to about to 5:01.50 to get an ok result.
In case, I could provide footage to join and sync if still helpful. 

Thanks a lot

Attached a screenshot showing the green double syncpoints and red the manualy added autosync point. 

[image]

AdrianEddy, 2022-12-07:
I added two more sync points now, please check https://gyroflow.xyz/devbuild/?autodownload
(please re-join the files)

AdrianEddy, 2022-12-07:
How about now?

helicocrasher, 2022-12-07:
Adding next segment (3 now) 
Stop the autosynch 
add 2s offset 
restart autosynch 
gives an OKish result except at the10min splice where one sync point receives a wrong offset --> 885ms.

[image]

In case there is a space where I can drop some GB of recordings I am happy to do so.

AdrianEddy, 2022-12-08:
I changed the sync defaults now, can you try again in latest dev build?

helicocrasher, 2022-12-08:
Massive improvement increasing the search size window default to 3s did the job - thank you
The majority of splice points get synced up good.
What can still occur:
- an inserted synch point is too near at the discontinuity of the gyro data at the splice and the resukting offset way off see picture above.
- Syncing up a lot of segments can start to fail towrds the end of the recording as the synch points seem to be located earlier with every splice point. Example recording with 10 segments fails at the beginning of the last segment.   The last syncpoint gets a wrong offset of -1753 ms instead of +2659ms (see pict below)

[image]

With all that said it already provides a lot of improvement, I would already use this build to save my significant manual work.
Thanks

helicocrasher, 2024-01-29:
Hello @AdrianEddy 

After a long while, I have been looking into the RunCam thumb Pro multi sements again.
The video segments are exactly 18'000 frames per 5 minutes segment (60FPS).  
The gyro data seems to be stored at 1kHz. 
Would require 300'000 Gyro lines per 5minute video segment.
The RunCam GCSV files contain: 
File 1. 301277 gyro lines 
File 2. 299'752 gyro lines 
File 3. 299743 gyro lines

The pattern is always the same the first GCSV file of a segmented recording is longer than the nominal 300'000 lines. 
The following files are about 250 to 300 lines (250-300ms) short.
To test I have extended the GCSV files with zero values to become 300'000 lines long, when the index (first value) is counted up, gyroflow can combine these extended GCSV files very reliable with almost zero time offset jump.
So no stretching of the gyro data is required. Stretching introduces shakes of 2 tow 3 seconds, filling in zeros is reducing the shakes to about 250 ms.
So thats a small update for now. I might try to motivate Runcam to fix the FW that missingg 250 to 300 gyro samples get also stored. 
Right now I am testing some c++ code to extend the GVCSV files to the corrrect length.      

[image]