gyroflow/gyroflow #512: Use gravity vectors displayed as enabled, but is disabled when loding new file

User-3090, 2022-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

876

### What operating system are you using?

Win10

### What GPU are you using?

3090

### What happened?

When horizon lock and use gravity vectors is enabled and a new file is dragged into Gyroflow the UI suggests that use gravity vectors is still enabled while in fact it is not. Please provide a way to make horizon lock and use gravity vectors a default via default.gyroflow.

User-3090, 2022-11-18:
Same thing with "Zooming center offset" sliders. They stay in the position set for the previous file, after opening a new file their value is neither reset nor applied. One needs to touch the sliders again to have their value applied.

I guess all those sliders should actually reset when loading a new file, but read and apply defaults from default.gyroflow.

AdrianEddy, 2022-11-19:
That was useful, fixed in 2b358e48a0d750f930809f1d0984b2827bcfb7fc

closed 2022-11-19 by AdrianEddy

User-3090, 2022-11-19:
Light speed dev, thank you so much for these incredibly fast fixes!