gyroflow/gyroflow #521: Crashes on virgl GPU

lattice0, 2022-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3 nightly

### What operating system are you using?

ubuntu 22.10

### What GPU are you using?

virgl on intel iris xe

### What happened?

crashes inside the virt-manager VM, virgl 3D renderer, after opening video. Don't know if it has anything to do with the virgl though,

### Relevant log output

```shell
lz@vm:~/Downloads/gyroflow$ ./Gyroflow-linux64.AppImage 
MDK 0.18.0 (git cef23ef) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang15.0.5 GCC4.2.1; libc++15005; 12:23:02 Nov 20 2022
failed to load libva.so.2, try libva.so.1
failed to load libva.so.1
failed to load libvdpau.so.1
failed to load libva.so.2, try libva.so.1
failed to load libva.so.1
mdk plugin 'mdk-braw': 0x55fc09f14a90
13:26:16 [ERROR] thread 'main' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/git/checkouts/ocl-4d61232778b4a85d/c7a0503/ocl/src/standard/platform.rs:50
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: main
  13: __libc_start_call_main
             at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  14: __libc_start_main_impl
             at ./csu/../csu/libc-start.c:381:3
  15: <unknown>

13:26:16 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
13:26:16 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "virgl", vendor: 65541, device: 0, device_type: Other, driver: "", driver_info: "", backend: Gl }
13:26:16 [WARN] Unknown GPU virgl
13:26:16 [DEBUG] (1) gyroflow::rendering: GPU type: Unknown, from name: virgl
13:26:19 [INFO] Latest version: v1.3.0, current version: 1.3.0 (gh899)
13:26:27 [ERROR] thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/git/checkouts/ocl-4d61232778b4a85d/c7a0503/ocl/src/standard/platform.rs:50
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: start_thread
             at ./nptl/./nptl/pthread_create.c:442:8
  14: __GI___clone3
             at ./misc/../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

13:26:27 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
13:26:27 [INFO] GPU list: ["[wgpu] llvmpipe (LLVM 15.0.2, 256 bits) (Vulkan)", "[wgpu] virgl (Gl)", "[wgpu] virgl (Gl)"]
13:26:28 [INFO] Lens profiles directory: "/tmp/.mount_Gyroflx8jpV6/camera_presets"
13:26:28 [INFO] Loaded 4974 lens profiles in 480.296ms
13:29:48 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/lz/Videos/ThumbPro_0000.MP4"
13:29:48 [DEBUG] (1) qml: Setting param smoothness 0.5
13:29:48 [DEBUG] (1) qml: Setting param smoothness_pitch 0.5
13:29:48 [DEBUG] (1) qml: Setting param smoothness_yaw 0.5
13:29:48 [DEBUG] (1) qml: Setting param smoothness_roll 0.5
13:29:48 [DEBUG] (1) qml: Setting param per_axis 0
13:29:48 [DEBUG] (1) qml: Setting param second_pass 1
13:29:48 [DEBUG] (1) qml: Setting param trim_range_only 0
13:29:48 [DEBUG] (1) qml: Setting param max_smoothness 1
13:29:48 [DEBUG] (1) qml: Setting param alpha_0_1s 0.1
13:29:48 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
13:29:48 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
13:29:48 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.video" ""
13:29:48 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
13:29:48 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
failed to load libva.so.2, try libva.so.1
failed to load libva.so.1
Crashdump written to /home/lz/Downloads/gyroflow/9bedcbe3-f939-4f19-401afbb2-67958ea8.dmp
Segmentation fault (core dumped)
```

AdrianEddy, 2022-11-21:
try .tar.gz package and run with `MDK_DECODERS=FFmpeg ./gyroflow`

But generally, Gyroflow is unlikely to work in a VM, it needs a normal GPU

AdrianEddy, 2022-11-23:
also you probably need to install libva2 and libvdpau1, and possibly other video/graphics packages, take a look at the [readme](https://github.com/gyroflow/gyroflow/#linux)

AdrianEddy, 2022-12-06:
I don't think it's a bug in Gyroflow. I think it's just missing libva2 and libvdpau1 packages, and even if that doesn't work, Gyroflow can run without any GPU acceleration:
- `export MDK_DECODERS=FFmpeg` will disable playback gpu decoding
- `export NO_OPENCL=1` will disable OpenCL
- `export NO_GPU=1` will disable wgpu

closed 2022-12-06 by AdrianEddy