gyroflow/gyroflow #522: optional "auto sync on load" (or don;t fall back to default settings)

nightflasher, 2022-11-23:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

This one is tricky to request, bc it's two in one.
i'm on gh899 and spotted this some nightlies (and in stable too) ago, the settings for the sync sections do no longer remain after restart and after loading a videofile (runcam thumb -pro in my case), the autosync starts immediately with parameters i don't like to use. 

So here's the request: 
- optional autosync (checkbox in ADVANCED)
- save/restore the last settings for synchronization (another checkbox?

i wrote a little tutorial and described "these settings will be always the last used ones", sure i could change this sentence - but i liked the behavior before and i think many like the new one, so an option is my request.

[image]
)

AdrianEddy, 2022-11-23:
The sync settings are set in lens profile, autosync as well. You can also create your own preset where you explicitly set "Do autosync". This is already implemented and available to use. The sync settings in lens profiles are hand-crafted to the particular camera and should work well. What exactly do you want to change and why?

Last sync settings are saved UNLESS they are loaded from lens profile, which take precedence because were manually tuned to that specific camera and conditions.

You can always edit the lens profile and update to your liking or just create a default preset

closed 2022-11-23 by AdrianEddy

AdrianEddy, 2022-11-23:
1. Sync search size: Do you ever have offsets larger than 1s in Runcam Thumb? Are you sure they are correct? that shouldn't happen
2. Essential matrix is not accurate, `rs-sync` with `OpenCV (DIS)` should be used
3. Low pass filter of 70Hz for OF data doesn't make sense, because it's at most video frame rate, ie. 30Hz or 60Hz.

What exact problems do you have with defaults? Can you send me some video+gyro file where defaults fail?

nightflasher, 2022-11-23:
1. yes, i have tested a multiple times and  my values work FOR ME better than the defaults
2. rs-sync change the gyro start to a later time, which can't be true for runcam thumb - video and gcsv are synced by the firmware!?
3. this gets rid of the vibrations
[image]

AdrianEddy, 2022-11-23:
1. Ok then we can slightly adjust defaults
2. It shouldn't, please send me this video sample, I'll figure out why
3. Definitely not, this is only for the chart (for drawing lines), doesn't affect stabilization

nightflasher, 2022-11-23:
https://www.dropbox.com/s/7oaxm5ucziemyi5/Thumb0020.zip?dl=0

AdrianEddy, 2022-11-23:
I'm getting good results with defaults and this file. 

First of all, you still need to estimate gyro bias for this camera, find a place where the drone doesn't move (flat lines), right click there and click "Estimate gyro bias here". Then autosync again.

Offsets definitely stay under 1s and rs-sync is doing a good job, but Runcam Thumb is a bad camera so to get acceptable results you'll still need to review the sync points, maybe remove one and add manually in a different spot. But this is unavoidable and the settings you proposed definitely don't produce better result than defaults + a quick review.

If you really want your settings you can modify `camera_presets\RunCam\RunCam_Thumb_1080p_16by9_1920x1080.json` in notepad and adjust settings there, but *I strongly discourage it* in favor of learning the process properly, to get best results.

You can also create a settings preset with your settings and save it in `camera_presets` as `default.gyroflow`

check out this project file: [Thumb0020.zip](https://github.com/gyroflow/gyroflow/files/10079069/Thumb0020.zip)
(open your video and then drag&drop this project file)

[image]