gyroflow/gyroflow #523: Rendering does not start - Gyroflow 1.3.0

matx93, 2022-11-23:
Gyroflow 1.3.0 does not start rendering, it is stuck at 0%
Synchronization seem to work ok in preview however when "Export" is launched, the rendering does not progress.
Running on W10 build 19042

BB file: https://1drv.ms/u/s!An6FA_wrlvnl109nLBkMZ6bxlBjB?e=XJwlFw 
Footage on Runcam Hybrid 2.7k 60fps: https://1drv.ms/u/s!An6FA_wrlvnl11B7LnUCLh6DZhjI?e=6y4N4G

[image]

AdrianEddy, 2022-11-23:
Close Gyroflow when this happens and upload `gyroflow.log` file, which is created next to Gyroflow.exe

matx93, 2022-11-23:
Unfortunately I can't see any log file :(

AdrianEddy, 2022-11-23:
try running Gyroflow as Administrator

matx93, 2022-11-23:
Log file available in attachment

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10079735/gyroflow.log)

AdrianEddy, 2022-11-23:
Can you also send me log from v1.2.0?
Do you have GPU encoding enabled?

matx93, 2022-11-23:
Gyroflow log v1.2.0 attached.
Unfortunately I have a quite old GPU (Nvidia GeForce 740M) which is not supported. 
GPU encoding is thus disabled.
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10079757/gyroflow.log)

AdrianEddy, 2022-11-23:
Can you try in v1.3.0 what happens if you select different `Device for video processing` under Advanced panel on bottom right?

matx93, 2022-11-23:
Ha! This actually made the trick. 
Here is the list of options under "Device for video processing":
[image]

It seems that OpenCL Intel HD Graphics 4000 does not work while the rendering starts and progresses with all the others. It is considerably slower with CPU only or with [wgpu] Microsoft Basic Render Driver Dx12 options.

[image]