gyroflow/gyroflow #526: Crashes on first time startup in select machines. Crash dump generated.

Abhro-sudo, 2022-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0

### What operating system are you using?

win11

### What GPU are you using?

nvidia gtx 1650

### What happened?

Startup fails. Works on my desktop. But on the laptop it crashes instantly and takes a dump, literally. 
[Gyroflow-windows64 (1).zip](https://github.com/gyroflow/gyroflow/files/10098847/Gyroflow-windows64.1.zip)

### Relevant log output

```shell
17:28:08 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
17:28:08 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.7.101") ["NVIDIA GeForce GTX 1650 Ti"]
Crashdump written to C:\Users\froy1\Downloads\Gyroflow-windows64 (1)/5eaa6e1e-b6e0-4ad8-968d-6ab4bd610ff0.dmp
Press any key to continue . . .
```

AdrianEddy, 2022-11-27:
Edit `Gyroflow_with_console.bat` and add `set NO_OPENCL=1` as a second line. Then run this bat file. If that works then it's an OpenCL error. OpenCL is included in your GPU drivers so make sure you have latest drivers installed. If you have two GPUs (common in laptops), make sure you update both GPU drivers.

Abhro-sudo, 2022-11-27:
That was it. Thank you so much. I should 'close as completed' right?

AdrianEddy, 2022-11-27:
Yes, there's nothing I can do about it on my side

closed 2022-11-27 by Abhro-sudo

AdrianEddy, 2022-11-27:
Most likely you need to update the Intel GPU driver for it to not crash with OpenCL

Abhro-sudo, 2022-11-27:
Understood. Thanks once again!