gyroflow/gyroflow #529: Quits unexpectedly when trying to open CinemaDNG compressed in SlimRaw.

toferzim, 2022-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0

### What operating system are you using?

MacOS 12.6

### What GPU are you using?

M1 Pro

### What happened?

It quits unexpectedly when I try to open a CinemaDNG sequence compressed in SlimRaw. The uncompressed sequence worked fine in GyroFlow. The compressed one works fine in Davinci Resolve.

toferzim, 2022-12-08:
Here is a short compressed sequence: https://disk.yandex.ru/d/8YZ5I73zGVX6cQ

AdrianEddy, 2022-12-08:
Thanks, it's a bug in ffmpeg, I reported it there with the sample file. There's nothing I can do in Gyroflow about it

closed 2022-12-08 by AdrianEddy