gyroflow/gyroflow #530: Quits unexpectedly when use AKAZE optical flow method.

toferzim, 2022-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0

### What operating system are you using?

MacOS 12&6

### What GPU are you using?

M1 Pro

### What happened?

GyroFlow quits unexpectedly when use AKAZE optical flow method with a CinemaDNG file. It works ok till 43% and the crashes. 

### Relevant log output

```shell
Can be found here: https://disk.yandex.ru/d/fLQVfXrtp5V-DQ
```

AdrianEddy, 2022-12-08:
AKAZE is very compute-intensive and RAM demanding, probably you ran out of memory. Can you try with lower Processing resolution? (Synchronization -> Advanced)
Also please include gyroflow.log

Also, AKAZE doesn't really have any advantage at this point, and will probably be removed in future versions

toferzim, 2022-12-08:
The thing is that any optical flow goes crazy when try to sync gyro data for a footage shot on a Helios 44-2 lens which is a 58mm lens. No idea why but it shows huge amplitude while the gyro data is  quite narrow.  In the same time with a 24mm lens optical flow works fine. I was thinking if AKAZE can do it better for a Helios footage.

AdrianEddy, 2022-12-08:
Latest dev build contains multiple `Pose method`s, for longer focal lengths `Almeida` seems to be better
https://gyroflow.xyz/devbuild/?autodownload

toferzim, 2022-12-11:
This built doesn't open my csv files( saying the format is not supported.

toferzim, 2022-12-11:
> The thing is that any optical flow goes crazy when try to sync gyro data for a footage shot on a Helios 44-2 lens which is a 58mm lens. No idea why but it shows huge amplitude while the gyro data is [...]

Even 480p doesn't work

AdrianEddy, 2022-12-11:
> This built doesn't open my csv files( saying the format is not supported. While the official build opens it just fine.
> 
> CSV is here: https://disk.yandex.ru/d/8YZ5I73zGVX6cQ

Rename to .gcsv

toferzim, 2022-12-11:
> > This built doesn't open my csv files( saying the format is not supported. While the official build opens it just fine.
> > CSV is here: https://disk.yandex.ru/d/8YZ5I73zGVX6cQ
> 
> Rename to .gcsv

Unfortunately neither this method of syncing is working.

AdrianEddy, 2022-12-11:
Can you share the footage, gyro and lens profile you're trying to stabilize?

toferzim, 2022-12-11:
Here it is:

https://disk.yandex.ru/d/8YZ5I73zGVX6cQ

toferzim, 2022-12-12:
Sorry^ have you managed to make it? By the way I've forgotten to mention the right orientation is ZXY

toferzim, 2022-12-13:
Well, if the built in Davinci stabilizer can stabilize can it mean it is possible to get an optical flow?  The right orientation is ZXY. It is fixed rigidly and 24mm footage with the same setup works ok with gyrofow. An offset is about 40mm from the sensor's center point up and same forward.

AdrianEddy, 2022-12-13:
I meant the time offset between gyro and video.
Davinci stabilizer works in a different way, it doesn't reconstruct the camera motion but simply moves pixels around in 2d.
The issue here is to reconstruct the physical camera motion, and DaVinci's stabilizer doesn't do that at all.

You can successfully stabilize this video is you have accurate time offset, optical flow is just a tool to calculate that offset

toferzim, 2022-12-16:
Unfortunately I have no idea about the offset, cause it is a different device with a separate trigger button. Seems like I need to think about syncing the start.

AdrianEddy, 2022-12-16:
If you just start recording together the offset should be within 1 sec, that's enough knowledge to then sync manually in Gyroflow

AdrianEddy, 2023-03-07:
I'm closing this because I don't know what else can be done to improve optical flow at large focal lengths, so there's nothing I can do here

closed 2023-03-07 by AdrianEddy